FreeOpcUa/python-opcua #334: UA Binary Doesn't Support GUID/UUID

zerox1212, 2016-10-17:
My project has GUIDs. I fixed importer to handle this, but it turns out ua_binary.py doesn't support UUID.

I starting to add this but it's kind of confusing.

Can someone help me out?

I tried this so far:
```python
class _Guid(_Primitive):

    @staticmethod
    def pack(uid):
        if uid is None:
            return Primitives.Int32.pack(-1)
        if isinstance(uid, uuid.UUID):
            uid = uid.bytes
        length = len(uid)
        return Primitives.Int32.pack(length) + uid

    @staticmethod
    def unpack(data):
        b = _Bytes.unpack(data)
        return uuid.UUID(bytes=b)
```

zerox1212, 2016-10-18:
So I looked at the spec and it says the Guid is a structure. Unfortunately the manual sucks at describing how to make structures in binary... the only example is to encode an extension object.

zerox1212, 2016-10-18:
After quite a bit of learning, I came up with this for ua_binary.py

```python
class _Guid(object):

    @staticmethod
    def pack(guid):
        # convert python UUID 6 field format to OPC UA 4 field format
        f1 = Primitives.UInt32.pack(guid.fields[0])
        f2 = Primitives.UInt16.pack(guid.fields[1])
        f3 = Primitives.UInt16.pack(guid.fields[2])
        f4a = Primitives.Byte.pack(guid.fields[3])
        f4b = Primitives.Byte.pack(guid.fields[4])
        f4c = bytearray(struct.unpack('8B', struct.pack('>Q', guid.fields[5]))[2:8])
        f4 = f4a+f4b+f4c

        # concat byte fields
        b = f1+f2+f3+f4

        return b

    @staticmethod
    def unpack(data):
        uuid_bytes = bytes(data.read(16))
        return uuid.UUID(bytes=uuid_bytes)
```

I didn't test unpack much yet (my guids are read only), but I think it will work.

zerox1212, 2016-10-18:
Can you show me where in the code guid is implemented? A node of type Guid was not working at all in the tests I did with UA Test Client. I hope I didn't waste a bunch of time writing the code above...

zerox1212, 2016-10-18:
Seems like I probably wasted my time... I now see Guid in `uatypes.py`. It appears the issue actually starts with the importer not creating NodeData correctly?

Should the Value of a GUID Variant be a string? or a python UUID object?
[image]

oroulet, 2016-10-18:
No time wasted, you probably learned a lot.

And guid implementation is probably broken since I have never used it (but
seems to have passed tests with other systems). I do not remember what the
value should be. A guid object in guess, but maybe a bytestring. The
importer might need special code for guid....

zerox1212, 2016-10-18:
How does Datetime Variant work? Does it get a python object?

oroulet, 2016-10-18:
I wrote special code to convert to datetime. Not sure we can do it for uuid
since there are many different types

zerox1212, 2016-10-18:
Spec shows this.
[image]

Does that mean Guid Variant value should be a hex string?

zerox1212, 2016-10-18:
After more testing, this also does not work:
'myguid = myobj.add_variable(idx, "MyGuid", ua.Variant('C496578A-0DFE-4b8f-870A-745238C6AEAE', ua.VariantType.Guid))'

Instantiating a Variant with a UUID also will not work because `guess_type` doesn't check for a UUID.

zerox1212, 2016-10-18:
Ok @oroulet,

We have 2 options to fix this.
1. The way I posted earlier: We treat Guid the same as datetime. The Varient Value is a python UUID object. Binary conversion is handled via Primitive. We delete the Guid object from `uatypes.py`. Finally we update Varient class's `guess_type` to include check for UUID. We then need to double check that NodeIds with Guid still work.
2. We change the Guid class to this:
```python
class Guid(FrozenClass):

    def __init__(self, value=None):
        if value is None:
            self.uuid = uuid.uuid4()
        else:
            self.uuid = uuid.UUID(value)
        self._freeze = True

    def to_binary(self):
        # convert python UUID 6 field format to OPC UA 4 field format
        guid = self.uuid
        f1 = uabin.Primitives.UInt32.pack(guid.fields[0])
        f2 = uabin.Primitives.UInt16.pack(guid.fields[1])
        f3 = uabin.Primitives.UInt16.pack(guid.fields[2])
        f4a = uabin.Primitives.Byte.pack(guid.fields[3])
        f4b = uabin.Primitives.Byte.pack(guid.fields[4])
        f4c = bytearray(struct.unpack('8B', struct.pack('>Q', guid.fields[5]))[2:8])  # no primitive .pack available
        f4 = f4a + f4b + f4c

        # concat byte fields
        b = f1 + f2 + f3 + f4
        
        # NOTE THAT self.uuid.bytes WILL NOT WORK FOR UA BINARY PROTOCOL; BYTE ORDER NOT CORRECT
        return b
```
This corrects guid packing and user can now create guid nodes:
```myguid = myobj.add_variable(idx, "MyGuid", ua.Variant(ua.Guid('C496578A-0DFE-4b8f-870A-745238C6AEAE'), ua.VariantType.Guid))```
One down side to this is that at the moment there is no way to `guess_type` when creating a Variant with a python UUID. It has to be a `ua.Guid()`.

Which option do you prefer?

zerox1212, 2016-10-21:
This issue is now fixed by https://github.com/FreeOpcUa/python-opcua/pull/338 . Note that instantiating nodes of type Guid is probably not handled currectly. Same goes for creating new nodes that have NodeId type Guid. Be sure to supply a your own `uuid.uuidx()` on creation.

closed 2016-10-21 by zerox1212