FreeOpcUa/python-opcua #350: Get Callback After Subscribe

zerox1212, 2016-11-02:
I would like to get a call back from the server after a list of nodes have been subscribed too.

My use case is that I am using internal subscriptions to keep my python objects syncronized with opc ua. The problem is that if I subscribe to a lot of nodes I have no way of knowing when they are all ready to be accessed.

Is there a way to get a callback from where ever the subscription is to tell the main thread that all the nodes are ready?

oroulet, 2016-11-04:
What do you mean by "all nodes are ready?"
Generally we could add a server API to get warning when a new subscription request is made. This could be used to limit number of subscriptions, add more logging, ...

zerox1212, 2016-11-04:
I think someone already added a subscription callback.

What I mean is that if you subscribe to 30 nodes for data changed, the subscription thread will call data_changed 30 times as it gets the value for each one. Is there a way to tell when this initial update has been completed?

zerox1212, 2016-11-15:
Can you answer this? Then I can close this issue.
>Or will the data_change callbacks be guaranteed to happen in the same order of subscription?