FreeOpcUa/python-opcua #351: Cache Not Working as Described

zerox1212, 2016-11-03:
@stanti 

In PR https://github.com/FreeOpcUa/python-opcua/pull/186 it says that if you give a path to a cache file it will be created automatically. When I look at your code this is not the case. The only way to get a cache file is if the user explicitly calls:
```python
def dump_address_space(self, path):
        """
        Dump current address space to path
        """
        self.aspace.dump(path)
```

Another thing to keep in mind is that you can't cache on the fly, because node objects with have objects that can't be serialized, such as thread locks from subscriptions.

zerox1212, 2016-11-03:
Also, this test:
`if cache_file and path.isfile(cache_file)` doesn't seem to work on windows, because if you pass the file name with no extension it returns false.
`path.isfile("myfile") = False`
`path.isfile("myfile.dat") = True`

How can I fix this in a nice way? I don't know if every OS makes the same 'shelve files'. 
On windows I get this:
[image]

oroulet, 2016-11-04:
This sounds like a documentation issue....

stanti, 2016-11-04:
When I created this issue my intention was to improve the startup performance on a raspberry pi.
The shelve file was created automatically on startup and just contained the standard opcua address space. All application specific nodes still had to be created dynamically to make sure that there are no serialization problems with various node objects. Furthermore changing the address space (especially deleting nodes) does not require a recreation of the whole cache when only the standard address space is cached.

However, I just saw that the automatic creation has been removed afterwards in PR #277 so you have to do the initial address space dump before you can make use of the cache (and the documentation should be updated as well).

I didn't know that shelving creates a multiple files on windows but I guess shelve.open will throw an exception if the shelve does not exist. So maybe catching the exception instead of  using path.isfile will work cross-platform.

oroulet, 2016-11-04:
Looks like I removed it when refactoring that part of the code. The test should be

    if cachefile is not None and not os.isfile(cachefile):
           dump
    else
           load

zerox1212, 2016-11-04:
Ok. I understand now.

I want to dump the entire address space, including my own nodes. The reason is that I was originally using a space dump and load to snapshot my server state.

Perhaps we should split these methods. 
1. for caching the standard address space
2. for dumping / loading the entire a space, for example if user wants to do an A space back up (like me)

zerox1212, 2016-11-04:
Now that I look at this closer I need help.
If you call dump when the address space is using the shelve, it will iterate over itself which currently raises NotImplemented.

How can I make calling `dump` multiple times function properly? It would need to only update the shelve.

Otherwise we can split caching and dumping...

zerox1212, 2016-11-04:
After a lot of investigation I think we need to split cache from dump. They do two different things.

I like the idea of the cache, but it needs to be finished. For example if you use the cache file right now you can't call `node.delete()` because the LazyLoaderDict doesn't support it. The cache probably breaks a lot of other things as well...

zerox1212, 2016-11-15:
https://github.com/FreeOpcUa/python-opcua/pull/353 is merged.

Original "caching" described by @stanti working as expected again. Standard dump and load are also working. Also added a few improvements to LazyLoadDict so that server functions such as deleting nodes works as expected. Read doc strings for full explanation.

Not that no server tests are done when a shelf file is supplied, this should be added at some point (run all server tests normal, then run all tests using a shelf).

closed 2016-11-15 by zerox1212