FreeOpcUa/python-opcua #352: Export of enumerations fails

bitkeeper, 2016-11-04:
`XmlImporter._value_to_etree`  treats an enumeration as an extension object:
```python
       if dtype.Identifier > 21:  # this is an extentionObject:
            _extobj_to_etree(el, type_name, dtype, val)
        else:
            val_el = Et.SubElement(el, "uax:" + type_name)
            _val_to_etree(val_el, dtype, val)
```
The compare doesn't work for custom datatypes (like a enum), also a compare for `>` makes no sence incase of if the nodeid is a `StringNodeId`.

Do you have a suggestion/direction howto how to fix this ?
Recursive search for the datatype until a Identifier in ns=0 is found and decide on that basis ?

Example output for an enummeration:
```xml
<UAVariable DataType="MyEnum" ParentNodeId="ns=1;i=5031" NodeId="ns=1;i=6012" BrowseName="2:State" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>MyEnum</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;s=5031</Reference>
        </References>
        <Value>
            <uax:Int32>0</uax:Int32>
        </Value>
</UAVariable>
```

oroulet, 2016-11-04:
Correct. I saw that code yesterday and wondered about it...

bitkeeper, 2016-11-07:
@oroulet 
Below is pseudo code that shows the current flow for exporting a simple variable:

As part of a fix I want to determine the basetype for the variable with a new function `get_variable_basetype `in `ua_utils`. It returns a nodeid with ns=0 and and i<=30.
Then a save compare can be made for i<=21.

For this fix I need to have either a node or server object in several of the helper functions like `_value_to_etree`, `member_to_etree` and `_val_to_etree`.

What is your prefered direction for this fix:

1. use as datatype a Node instead of a NodeId 
2. pass an additional server argument
3. make the helper methods part of XmlExporter
4. ...

Pseudo code of the variable export flow:
```python
def XmlExporter.add_variable_common       
  value_to_etree      # simple value flow 

def value_to_etree
  _value_to_etree     # simple value flow
    
def _value_to_etree
  if                   # problem is here
    _extobj_to_etree
  else
    _val_to_etree  # simple value flow
  
def  _val_to_etree
  if
                  # simple value flow ends here
  else
    for
      member_to_etree

def _extobj_to_etree
  for 
    member_to_etree

def member_to_etree
  if
    for 
      _value_to_etree
  else
    _val_to_etree
```

oroulet, 2016-11-07:
Hi,
This is method is supposed to already exist in ua_utils.py. It is called
dtype_to_vtype()  maybe we should rename it without abbreviation. So in
theri we can just call it to get correct variant type
Please add a test, we need to make the xml import /export reliable.

oroulet, 2016-11-07:
The xml exporter has already a pointer to server object, so this should be straightforward

bitkeeper, 2016-11-07:
The xmlexport indeed has a pointer to the server object, but the used helper methods in xmlepxort.py not (are just plain methods, not class methods).

oroulet, 2016-11-07:
Then it is a good argument to move these functions into class.

zerox1212, 2016-11-07:
I don't recall a reason why they can't be moved to the class. They are only used by exporter.

closed 2016-11-08 by bitkeeper