FreeOpcUa/python-opcua #360: tests_server.TestServerCaching fails with 0.90.0

MartinBorg, 2016-11-12:
This is on Debian sid with Python 3.5.1
(I'm not sure whether the tests are supposed to work with Python 3)
```
======================================================================
FAIL: runTest (tests_server.TestServerCaching)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/build/python-opcua-0.90.0/tests/tests_server.py", line 470, in runTest
    self.assertEqual(server.get_node(id).get_value(), 123)
AssertionError: None != 123

----------------------------------------------------------------------
```

oroulet, 2016-11-12:
Tests should ask run fine. And they do on my PC and on Travis. But this one is interesting as it also happens on scrutinizer

zerox1212, 2016-11-14:
Shelve seems to do different things depending on the platform. It could also be a file permission issue.