FreeOpcUa/python-opcua #361: suspicious test output for three tests with 0.90.0

MartinBorg, 2016-11-12:
This is on Debian sid with Python 2.7.11.
The tests do not fail, but I wonder why I see those exceptions,
because the tests do not look like exceptions tests.
```
test_method (tests_client.TestClient) ... ERROR:opcua.server.address_space:Error executing method call CallMethodRequest(ObjectId:TwoByteNodeId(i=85), MethodId:StringNodeId(ns=2;s=ServerMethod), InputArguments:[Variant(val:2.1,type:VariantType.Double), Variant(val:89,type:VariantType.Int64), Variant(val:9,type:VariantType.Int64)]), an exception was raised: 
Traceback (most recent call last):
  File "./opcua/server/address_space.py", line 424, in _call
    res.OutputArguments = node.call(method.ObjectId, *method.InputArguments)
  File "./opcua/common/methods.py", line 57, in wrapper
    result = func(parent, *[arg.Value for arg in args])
TypeError: func() takes exactly 2 arguments (4 given)
ok
test_historize_variable (tests_server.TestServer) ... ERROR:trollius:Exception in callback _sub_loop()
handle: <TimerHandle when=55346.236289 _sub_loop()>
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/trollius/events.py", line 136, in _run
    self._callback(*self._args)
  File "./opcua/server/internal_subscription.py", line 284, in _sub_loop
    self.publish_results()
  File "./opcua/server/internal_subscription.py", line 312, in publish_results
    self.callback(result)
  File "./opcua/common/subscription.py", line 113, in publish_callback
    self._call_event(notif)
  File "./opcua/common/subscription.py", line 149, in _call_event
    data = self._monitoreditems_map[event.ClientHandle]
KeyError: 201
ok
test_method (tests_server.TestServer) ... ERROR:opcua.server.address_space:Error executing method call CallMethodRequest(ObjectId:TwoByteNodeId(i=85), MethodId:StringNodeId(ns=2;s=ServerMethod), InputArguments:[Variant(val:2.1,type:VariantType.Double), Variant(val:89,type:VariantType.Int64), Variant(val:9,type:VariantType.Int64)]), an exception was raised: 
Traceback (most recent call last):
  File "./opcua/server/address_space.py", line 424, in _call
    res.OutputArguments = node.call(method.ObjectId, *method.InputArguments)
  File "./opcua/common/methods.py", line 57, in wrapper
    result = func(parent, *[arg.Value for arg in args])
TypeError: func() takes exactly 2 arguments (4 given)
ok
```

oroulet, 2016-11-12:
This is normal. We test for crash

oroulet, 2016-11-12:
Maybe the key error stuff is not normal... Not sure