FreeOpcUa/python-opcua #362: The xml parser crash when loading a file with a Models tag

levkevi, 2016-11-15:
When trying to load a file with a Models tag, the parser crash.

Here's an example file with the problematic tag.

[models_are_problematic.txt](https://github.com/FreeOpcUa/python-opcua/files/593234/models_are_problematic.txt)

levkevi, 2016-11-15:
It should be fixed in #363

zerox1212, 2016-11-15:
Thanks. Do you know if there is a list of "non-node" XML tags in the spec somewhere?

oroulet, 2016-11-16:
What are model tags? does anyone have documentation for that? Maybe this is where we should save xml files used as references? would be great for modeler

levkevi, 2016-11-16:
I don't know but I compiled a data model file to a node file using this tool:

https://github.com/OPCFoundation/UA-ModelCompiler

You might find something looking at the code.

oroulet, 2016-11-16:
What is a data model file?

zerox1212, 2016-11-16:
I think it's this:
https://github.com/OPCFoundation/UA-ModelCompiler/blob/master/ModelCompiler/UA%20Model%20Design.xsd

Example:
https://github.com/OPCFoundation/UA-.NET/blob/master/SampleApplications/Samples/Common/Sample/SampleDesign.xml

levkevi, 2016-11-16:
I can confirm, this is the file that @zerox1212 linked.

zerox1212, 2016-11-16:
Looks like OPC Foundation uses this model format to generate nodes and code (classes). Seems like a different approach than python modeler.

oroulet, 2016-11-17:
What do you mean? The xsd defines the xml format. Is there another file
somewhere?

bitkeeper, 2016-11-17:
Hard to find some information:
- The only real use of UAModelDesign so far looks in the OPCFoundation ModelCompiler  (and even that tool generates also  UANodeSet as output)
- Other tools use UANodeSet
- In https://opcfoundation.org/UA/schemas/1.02/UANodeSet.xsd the Models element is missing and in newer https://opcfoundation.org/UA/2011/03/UANodeSet.xsd it is present.
- Unknown if there are any tool for designing the model based on  UAModelDesign ? 

@levkevi what do you mean with 'you usually define your models ...' ? By hand or tool ?

Looks like the UAModelDesign is more 'object-oriented' approach of describing the namespace. More easy to parse; object/objectype contain its members as nested childs, instead of nodeid refs to other nodes.

oroulet, 2016-11-17:
Does it specify nodeids too? I could not find any , and I am not an expert in xsd...

levkevi, 2016-11-17:
@bitkeeper 
I was told that it was typed by hand.

@oroulet 
It doesn't specify the nodeid, you have to compile it with their tool (the one linked in their repo) to generate the Node file.