FreeOpcUa/python-opcua #365: Client write callback in server

joaopmrod, 2016-11-16:
How do I trigger a handle function in the server when a client writes to a variable?

oroulet, 2016-11-17:
You subscribe to the variable on server side?

joaopmrod, 2016-11-17:
But then I will get a callback even when I write to it...

oroulet, 2016-11-17:
Yes. Do you have a good use case for only tracking writes from clients?

joaopmrod, 2016-11-17:
I'm writting a OPC server for a specific instrument. When a value changes in the instrument I want to update it in the OPC server and do nothing else. 

When a client changes is in the OPC server I want to trigger a callback function that will change that varible in the instrument.

joaopmrod, 2016-11-17:
Can I have a callback in the server when a client try to read one node?

oroulet, 2016-11-17:
If you want to monitor everything handing in server you can override the
InternalServer class and the session class then you can know everything
happening. But I am suspecting a bad design in your application if you need
such things....

zerox1212, 2016-11-17:
You could just keep track of values being written by the instrument. Then the subscription call back from a server write will always have the same value. If a client writes the subscription call back will have a different value and you will know it wasn't from the server it self.

oroulet, 2016-11-18:
But I am interested to know in what context, we have a value from an
instrument, which should be written by clients. That does not make sense in
my mind.

zerox1212, 2016-11-18:
My guess is that he is reading a parameter cyclically from the instrument and publishing it on the server. Then he wants to catch if an external client writes to the same address so he can change the instrument parameter.

joaopmrod, 2016-11-18:
Exactly

oroulet, 2016-11-18:
That does not explain anything! But thinking about it there might be cases
where one would like not to generate events. For example if the underlying
system has another interface that modify its variables, one probably want
to sync to changes from that interface without generating events to avoid
update loops.
To support that we need to add one more argument to Node.set_ value() and
server. write(), called for example no_event=False.

oroulet, 2016-11-18:
Maybe it is easier to add a set _ silent() method to the session in
internal server. Then we do not need to modify api

zerox1212, 2016-11-18:
I don't understand what you mean.

He basically wants to subscribe to a node for data change notification, but ignore all data changes from the server itself (internal server/session).

bitkeeper, 2016-11-21:
Isn't this the same request as #310 ?

The internal session already knows if the set value comes from external or internal:
```python
 def InternalSession.write(self, params):
        if not self.external:
            # If session is internal we need to store a copy og object, not a reference,
            # otherwise users may change it and we will not generate expected events
            params.NodesToWrite = [deepcopy(ntw) for ntw in params.NodesToWrite]
        return self.iserver.attribute_service.write(params, self.user)
```
Can we not provide self.external to the `iserver.attribute_service.write` and the to the `address_space.write` ?
Then the address_space.set_attribute_value could be modified not to called the subscription when it concerns an internal session ?

oroulet, 2016-11-21:
Exactly. That is why I proposed to add a method called set _ silent or set
_ no_ data _ change to the session class on server side. The alternative is
to add extra argument to set_ attribute _ value, set_ value, etc...

EnzoOestanto, 2018-01-17:
hi, i would like to ask  on how to write a value from a client.
i tried "client.set_value(var,0);" and "client.write_value" both of the resulted in a error message that there is no attribute called set_value or write_value. what is the correct attribute name?

zerox1212, 2018-01-17:
Read the docs. The functions you want are on the `Node` class. http://python-opcua.readthedocs.io/en/latest/node.html

closed 2022-09-01 by joaopmrod