FreeOpcUa/python-opcua #375: Set Bad Quality to a Tag

joaopmrod, 2016-11-22:
I'm trying to write a Bad Quality to a tag. I'm doing this:

a =ua.uatypes.DataValue()
a.StatusCode=ua.uatypes.StatusCode(301)
self.opcNode.set_value(a)

However, when I read the tag from a client it always has good quality? 

How can I set the tag quality to bad quality?

simone201, 2017-01-04:
I'm trying to use
node.set_value_rank(StatusCodes.Bad)

and i can't get it working too.
Any help would be appreciated from the devs!

zerox1212, 2017-01-04:
Not sure without stepping through the code, but my guess it that since the write was successful there is a call to ua.StatusCode() someplace that is overwriting your status.

simone201, 2017-01-05:
Strange thing is that calling "get_value_rank" returns the right set value at any time during execution, while in the OPC Client the rank is always set to Good.
Any other ideas or ways to set the quality of a node?

oroulet, 2017-01-05:
I did exactly what you did and a client (uaexpert for example) receive the bad status code.
our client in fact raises an exception since it checks the returned code

oroulet, 2017-01-05:
value ranks has nothing to do with status code or quality, isnt'it?

closed 2017-01-07 by oroulet