FreeOpcUa/python-opcua #377: Parsing "ListOfXXX" in XML

drakes00, 2016-11-28:
Hi, 

I stumble across an XML file that cannot be parsed.
It seems to be related to it containing ListOfDateTime, ListOfGuid and ListOfQualifedName.
The case of DateTime does not seem too hard and I tried adding this piece of code to ua_type_to_python from xmlparser.py:

```
    elif uatype in ("DateTime"):
        # Format specified in Part6 Mappings: 5.3.1.6 DateTime
        try:
            # Zulu datetime
            return datetime.datetime.strptime(val, "%Y-%m-%dT%XZ")
        except ValueError:
            try:
                # Timezone offset.
                return datetime.datetime.strptime(val, "%Y-%m-%dT%X%Z")
            except ValueError:
                raise Exception("DateTime object does not meet specifications's format: ", val)
```
I am not sure if its enough but its at least seems to do the trick in my case.

However, in case of Guid and QualifiedName the `val` argument of the function comes empty and I'm not sure why.

Also, still in the same function type "Bytes" appears twice in the list (most likely one is "Byte" without an s) and SByte seems missing.

I proposed a pull request #376 for Bytes type and DateTime, however I feel like I am duplicating code from _parse_value, am I wrong?
It seems to me that one is used to parse single values and the other to parse lists, if so maybe a more unified parsing of "ListOf" and not "ListOf" values would be better.

Cheers,
Maxime

zerox1212, 2016-11-28:
We need to ask @oroulet because I'm not sure why "ListOf" uses `ua_type_to_python` but other values that have native time are in `_parse_value`. I know GUID is working if it is NOT in a list, it gets parsed with this code in `_parse_value`.
```python
elif ntag == "Guid":
            self._parse_value(child_el, obj)
            obj.valuetype = obj.datatype  # override parsed string type to guid
```
However this won't get called if it's in a list. I have never used lists and the XML code is quite complicated.

oroulet, 2016-11-28:
yes you are right there is some duplication here. We could call ua_type_to_python in _parse_value at several places. Can you fix this since you are looking at it? this may have small consequences in xml_importer.py

and for converting a string to a datetime, there is probably allready a function in uatypes.py. Can you have a look?

oroulet, 2016-11-28:
wrong the function does not exist. but look in ua_utils.py. there are function string_to_val and val_to_string that duplicate a lot of things, maybe we should merge this methods....
we can make ua_type_to_python construct a VariantType and use string_to_val

drakes00, 2016-11-28:
Ok, I'll look at those and see what I can do to unify things.

oroulet, 2016-11-29:
I know that the xml code is quite hard to debug (propositions to improve it welcome) so do not try to do difficult things, but use ua_type_python in _parse_value and rewrite ua_type_to_python to something like

    def ua_type_to_python(val, uatype_str):
          return string_to_val(val, getattr(ua.VariantType, uatype_str))

oroulet, 2016-11-29:
function could be rename to ua_primitive_to_python to make it clearer what it does

drakes00, 2016-12-02:
Hum the use of string_to_val may have introduced a bug when parsing ByteString because it encodes it in utf8 and then tries to call replace on it (which fails in my setup).
I tried to comment the encode in string_to_val and it seems fine.
I added test cases for this type also and they pass.
Why is this encoding important?

oroulet, 2016-12-03:
@drakes00 I never remember what the correct syntax is but in that method we need to convert a string to bytes. There are many ways to do it but only one (or a few) work with python 2 and 3 so we need to test for both.
"lkjklj".encode("utf-8") seems to work for both python version. When do you get an error?

drakes00, 2016-12-05:
At line 237 of xmlparser.py:

```
elif ntag in ("ByteString", "String"):
    mytext = ua_type_to_python(val_el.text, ntag)
    if mytext is None:
        # Support importing null strings.
        mytext = ""
    mytext = mytext.replace('\n', '').replace('\r', '')
    obj.value = mytext
```

As mytext is encoded to utf8, we should not be able to call replace on it in python3.

You can test it with:
```
python -c '"aaa".encode("utf8").replace("a", "b")'
python3 -c '"aaa".encode("utf8").replace("a", "b")'
TypeError: 'str' does not support the buffer interface```

drakes00, 2016-12-20:
I guess this can be closed now.

Cheers,
Maxime

closed 2016-12-20 by drakes00