FreeOpcUa/python-opcua #385: Fail to read more than 50 Tags in python-opcua client application with TIA Portal OPC UA Server simulator

bhoomil-avch, 2016-12-13:
Hi,
I have found one issue with python-opcua client APIs.
Please go through the following link as I already discussed this bug on FreeOpcUa Google Group,
[https://groups.google.com/forum/#!msg/freeopcua/r-OasG4eshg/rhWSmxplBQAJ](url)

The issue is in the following call stack.
[image]

zerox1212, 2016-12-13:
As a possible workaround you could see if TIA Portal UA server has a setting to turn off truncating browse results.

bhoomil-avch, 2016-12-15:
I'm new at TIA Portal currently,
till now I don't find anythings regarding "turn off truncation"
I'm surprise that why uaexpert.exe gives me the perfect result.
As you said that go with workaround by turning off truncation off at server side but are there any workaround possible at client side? because ultimately problem is in the client APIs

oroulet, 2016-12-15:
Python client must be fixed as explained in mailing list. This is not
difficult

oroulet, 2016-12-19:
https://github.com/FreeOpcUa/python-opcua/pull/386

bhoomil-avch, 2016-12-20:
Thanks.,, the changes you have done is now working fine.

closed 2017-01-07 by oroulet