FreeOpcUa/python-opcua #391: Failed to install OPC-UA GUI client.

praful0177, 2017-01-07:
Hi,
I'm trying to install OPC-UA GUI client using the command _**"pip install freeopcua-client"**_ on windows 7 PC ,for this getting an error as below:
Python version -3.5.2
Error:
">pip install freeopcua-client
Collecting freeopcua-client
  Could not find a version that satisfies the requirement freeopcua-client (from
 versions: )
No matching distribution found for freeopcua-client."

I have already installed the required utilities as mentioned in readme like cryptography, dateutil and pytz an d winpython .
Is this a real issue or am I missing anything?

Thanks
Praful

oroulet, 2017-01-07:
I removed the outdated package. The new one is called opcua-client . Where
did you find mention of freeopcua-client? We should fix it

praful0177, 2017-01-07:
Thank you , I was able to install gui-client application using the command "pip install opcua-client".
I find the incorrect one on https://github.com/FreeOpcUa/opcua-client-gui under How to Install steps.
Is there any other webpage or documentation freeopc maintains, where I can find the updated procedure, if so can you please share the same?

Regards
Praful

praful0177, 2017-01-07:
As per the (3) step provided for windows, 
3.Run via the script pip created: YOUR_INSTALL_PATH\Python\Python35\Scripts\o**pcua-modeler.exe**
The mentioned binary is not available, so I ran "opcua-client.exe" from command prompt, but getting an error as 
"te-packages\uaclient\mainwindow.py", line 9, in <module>
    from PyQt5.QtCore import pyqtSignal, QTimer, Qt, QObject, QSettings, QModelI
ndex, QMimeData, QCoreApplication
ImportError: No module named 'PyQt5' "
Full error log is attached.
Any suggestion?

Thanks
Praful
[opcua_client_gui_error_log.txt](https://github.com/FreeOpcUa/python-opcua/files/691547/opcua_client_gui_error_log.txt)

oroulet, 2017-01-07:
thanks for the report. I fixed the README

oroulet, 2017-01-07:
>>> mportError: No module named 'PyQt5' "

you need to install the version of winpython with pyqt5. or install pyqt5
yourself

closed 2017-01-07 by oroulet