FreeOpcUa/python-opcua #392: Client certificates are not validated

coolmast, 2017-01-11:
When using certificates for authentication at the OPC UA server, the python-opcua stack at no point validates the certificate of the client. It is even possible to provide mismatched private keys to the certificate which underlines that there is no check at all for the validity of the certificate.

oroulet, 2017-01-11:
This is true. Even more we do not check if a secure connection is established, do not check message ids, etc,... When I first developed the stack my goal was to connect to anything, not to be percky with servers. But patches are welcome to improve the situation...

zerox1212, 2017-01-11:
I thought the keys and certs were mainly for encryption not authentication?