FreeOpcUa/python-opcua #401: OPC Client user authentication failed with '@' in the password

Henna1977, 2017-02-06:
Hello,

I got an issue with the authentication of a user on an OPC Server (Siemens TIA PLC).
The password contains '@' so I get an error when I try to connect to the OPC Server.

errno: 11001
strerror: 'getaddrinfo failed'

The server URL is like this:
'opc.tcp://user:P@ssword@xxx.xxx.xxx.xxx:4840/'

Is there an other way to connect with username and password?
In the example code I only find the solution by sending the user and password in the URI.

Best regards
Henna

AlexejStukov, 2017-02-07:
I think the best approach would simply be: don't use `/,\,@,:` in your password, username or url...

oroulet, 2017-02-07:
The low level interface is always available. Look at the code connect() method, find out where the password is set (it might need to be encrypted) and set it directly.
But I agree that we should add methods set_password and set_user, iin addition to using url. It would also be usefull for GUI. help is welcome.

Henna1977, 2017-02-07:
>Thx
I take a look and will make a pull request for the two methods.

closed 2017-02-07 by Henna1977