FreeOpcUa/python-opcua #410: node.getChildren() max 100 children?

maiklof, 2017-02-27:
Hi,
First of all, thanks for this great module! 
I am just having some issues getting the children of a node which has more than 100 variables. I can see all the variables using UAExpert client, but python-opcua seems to return just the first 100 variables. Is it possible? 
The functions says "Get all children of a node. By default hierarchical references and all node classes are returned.". I call it without any parameter, so I would get all the variables, right?
Thanks!
Mikel

oroulet, 2017-02-27:
what version are you using?
We fixed this in https://github.com/FreeOpcUa/python-opcua/commit/dbd124928e96fa34888290affdcf93322226e8a6
maybe it is not released yet... I should do a new release...

oroulet, 2017-02-27:
new released to pip done

maiklof, 2017-02-27:
Ok, because, just to be sure, I downloaded the latest version from Gihub this morning before posting the issue...
But now I have download the latest with 'pip freeopcua --upgrade' and it works!
Is it any difference between downloading the source in zip format and installing vs doing pip?
Nice work anyway!
Mikel

zerox1212, 2017-02-27:
The master on Github tends to get ahead of the pip releases. If you took the latest code from here then either a bug has been introduced, or maybe you weren't on the latest commit to master.