FreeOpcUa/python-opcua #414: Custom event properties aren't send in example server-events

bitkeeper, 2017-03-03:
While fixing an exception in `examples/server-events.py` in PR #413 , I noticed that in UaExpert not all properties of the custom event where received/transmitted.

Should the following fragement of the example work ?:
```python
etype = server.create_custom_event_type(idx, 
                                              'MyFirstEvent', 
                                              ua.ObjectIds.BaseEventType,
                                              [('MyNumericProperty', ua.VariantType.Float), 
                                               ('MyStringProperty', ua.VariantType.String)])
evgen = server.get_event_generator(etype, myobj)
...
evgen.event.Message = ua.LocalizedText("MyFirstEvent %d" % count)
evgen.event.Severity = count
evgen.event.MyNumericProperty = count
evgen.event.MyStringProperty = "Property " + str(count)
evgen.trigger()
```

Output of UaExpert:

Both Events are received:
[image]

Looks like only the BaseEventType properties are send:
[image]

oroulet, 2017-03-03:
uaexpert does not check what properties an event has and only ask for the properties of BaseEventType. It might be that it does not manage to find the event definition, but I do not think so. It is a limitation of uaexpert

bitkeeper, 2017-03-03:
Found it, just my fault! 
You have to configure the UaExpert to show more then the basic event events:

- In the UaExpert configuration of the EventView 
- Click on '>'  at SimpleEvents 
- Enable your own event type or one or more properties of your event (or just select all at the checkbox at SimpleEvents)

[image]

closed 2017-03-03 by bitkeeper

oroulet, 2017-03-06:
thanks for the info