FreeOpcUa/python-opcua #415: I cant suscribe to datachagne at the Server side

MRScreen, 2017-03-04:
Hello everyone,

I am not really sure if i am doing right, but I have taked a look and I think that the way I am doing make sense. However it is not working, I am getting an exception.

My code is:

self.handler = SubHandler(opc_tag = self, logger = self.logger)
sub = server.create_subscription(100, self.handler)
self.handle = sub.subscribe_data_change(self.tag_object)

And the exeception i get when it is calling server.create_subscription(...)
  File "...\site-packages\opcua\common\utils.py", line 162, in call_later
    p = functools.partial(self.loop.call_later, delay, callback)
AttributeError: 'NoneType' object has no attribute 'call_later'

Am I doing it right, or it is a bug? Because I have debug to that point (internal_subscriptino.py, funtion: _subsctiption_loop) and the atribute service.loop has no attribute call_later 

    def _subscription_loop(self):
        if not self._stopev:
            self.subservice.loop.call_later(self.data.RevisedPublishingInterval / 1000.0, self._sub_loop)

Thanks!

MRScreen, 2017-03-04:
I have forget to say that I am using 3.5.2

oroulet, 2017-03-04:
Have you started the server?

MRScreen, 2017-03-04:
Not yet, I just have realized that the ThreadLoop is been created on run. So, should I start the server before subscribing?

MRScreen, 2017-03-04:
Sorry I was wrong, I can start subscribing after start the server.

Thanks a lot.

closed 2017-03-04 by MRScreen