FreeOpcUa/python-opcua #417: example/server-event.py

aquari89, 2017-03-06:
Hello, All
I have been interested in OPC UA a few days ago and am studying through the python-opcua example.

But I have something to ask.
I have tried many, but I do not know which part to touch.
That is status_change Notification

Anybody tried to use status_change Notification?
If you have any related methods or example code, I would appreciate it if you could teach me.

I want to implement it as shown in the following figure.
Call the method so that we can see the state change as an event.
[image]

oroulet, 2017-03-06:
I implemented it while implementing the other types of events, but I am not sure anybody has ever used it, there are no tests, and there might be bugs..... if it does not work, you might want to enable debugging and add prints to find out...

aquari89, 2017-03-06:
@oroulet Thank you for your reply!
Can you tell me the link address that you have done for reference?

oroulet, 2017-03-06:
server or client?

aquari89, 2017-03-06:
@oroulet server!!

oroulet, 2017-03-06:
Reading again you description, I am unsure about what you want to do. StatusChancge events are for server events. A server is suppose to send events when starting and stopping. (This is by the way not implemented)
But it looks like what you want to is simply send a custom event when something happens in your system. This is working. See for example: https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-events.py

aquari89, 2017-03-06:
@oroulet 
For example, when I call 'Start' method, 'machineState' variable sets True.
And then, I call 'Stop' method, 'machineState' variable changes False.
At this point, it is my project's intention to raise an event that the value has changed. (As you can see from the picture above)

However, with the 'examples/server-events.py' sample code,  I can not implement the event I want.

oroulet, 2017-03-06:
I you all you want to do is to monitor a variable then subscribing to data change is all you need. Events are usufull when you want to be sure you do want to miss en event, while data is usually what you want to monitor a state or value.

aquari89, 2017-03-06:
@oroulet 
Thank you for your advice.
I'll give it a try.

aquari89, 2017-03-07:
@oroulet
I tried a lot, but it does not work well.
Is the monitored item generated from the node?
How can FreeOpcUa(python) monitor variables and subscribe to data changes?

Here is my example code.
How do I program an event that has a value change after calling a method?

```
import sys
sys.path.insert(0, "..")
import time, random
from datetime import datetime
from opcua import ua, uamethod, Server

@uamethod
def Start(parent):
    stateV.set_value(True)
    print 'Machine starts!'

@uamethod
def Stop(parent):
    stateV.set_value(False)
    print 'Machine stops!'

if __name__ == "__main__":

    server = Server()
    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")

    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)
	

    objects = server.get_objects_node()
    machine = objects.add_object(idx, "Machine")

    stateV = machine.add_variable(idx, "machineState", False)
    temperatureV = machine.add_variable(idx, "machineTemperature", 25.0)

    startM = machine.add_method(idx, "Start", Start)
    stopM = machine.add_method(idx, "Stop", Stop)

    server.start()

    try:

        while True:
            time.sleep(1)

            if stateV.get_value() == True:
                temperature = random.uniform(-10.0, 25.0)
                temperatureV.set_value(temperature, varianttype = ua.VariantType.Double)
                temperature = 0.0

    finally:
        server.stop()
```

zerox1212, 2017-03-07:
So I looked at your problem and found a problem in python opcua:
There is no way to get a reference to the EventGenerator object if you are browsing from the Source Object.

Because of this the only way I could solve your issue is with a hack using `global`.

Make your function trigger your objects event generator via the "hack" function:
```python
@uamethod
def Start(parent):
    stateV.set_value(True)
    print 'Machine starts!'
    method_event("machine starts", 500)  # this will generate the UA event
```

Add the hack function:
```python
def method_event(message, severity):
    global machine_evgen
    machine_evgen.event.Message = ua.LocalizedText(message)
    machine_evgen.event.Severity = severity

    machine_evgen.trigger()
```

Finally, create the an event generator for your object (add this line to your server code):
```python
machine_evgen = server.get_event_generator(ua.ObjectIds.BaseEventType, machine)
```

@oroulet If a node has "GeneratesEvents", how can you get a reference to the EventGenerator?

aquari89, 2017-03-08:
@zerox1212 
Thank you!
It works well.

By the way, I have a question.
In "example/server-event.py" code, the code creates a custom event type and add a property. 
I don't know why the property is there and what it does.

More specifically,
For example, Suppose I create a custom event type called "Machine_control_EventType" and add the 'state' property and the 'temperature' property.
` etype = server.create_custom_event_type(idx, 'Machine_control_EventType', ua.ObjectIds.BaseEventType, [('machineTemperature', ua.VariantType.Double), ('machineState', ua.VariantType.Boolean)])`

I am wondering how the 'machineState' variable and the 'machineTemperature' variable created in my code are related to the properties of custom event types and how they function in terms of events.
`    stateV = machine.add_variable(idx, "machineState", False)
    temperatureV = machine.add_variable(idx, "machineTemperature", 25.0)`

destogl, 2017-03-08:
Regarding custom event types, they will extend one of the build-in types (usutally BaseEventType) and add your properties there. When you trigger event you can fill this two properties with data and transfer some information with the event.

They are **not** in any way connected to the properties of your object.

aquari89, 2017-03-08:
@destogl 
I'm sorry but I can't understand exactly your words.("When you trigger event you can fill this two properties with data and transfer some information with the event.")

Could you explain it through the code at the bottom?

For example, when an event is triggered, does it mean that the machineTemperature property and the machineState property are set to the changed data value?
Also, What does mean "transfer some information"? 
I wonder what kind of information you say.

`import sys
sys.path.insert(0, "..")
import time, random
from datetime import datetime
from opcua import ua, uamethod, Server

def method_event(message, severity):
    global machine_evgen
    machine_evgen.event.Message = ua.LocalizedText(message)
    machine_evgen.event.Severity = severity

    machine_evgen.trigger()

@uamethod
def Start(parent):
    stateV.set_value(True)
    print 'Machine starts!'
    method_event("machine starts", 111)

@uamethod
def Stop(parent):
    stateV.set_value(False)
    print 'Machine stops!'
    method_event("machine starts", 222)

if __name__ == "__main__":

    server = Server()
    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")

    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)
	

    objects = server.get_objects_node()
    machine = objects.add_object(idx, "Machine")

    temperatureV = machine.add_variable(idx, "machineTemperature", 25.0)
    stateV = machine.add_variable(idx, "machineState", False)

    startM = machine.add_method(idx, "Start", Start)
    stopM = machine.add_method(idx, "Stop", Stop)

    etype = server.create_custom_event_type(idx, 'Machine_control_EventType', ua.ObjectIds.BaseEventType, [('machineTemperature', ua.VariantType.Double), ('machineState', ua.VariantType.Boolean)])

    evgen = server.get_event_generator(etype, machine)

    machine_evgen = server.get_event_generator(ua.ObjectIds.BaseEventType, machine)

    server.start()

    try:

        while True:
            time.sleep(1)

            if stateV.get_value() == True:
                temperature = random.uniform(-10.0, 25.0)
                temperatureV.set_value(temperature, varianttype = ua.VariantType.Double)
                temperature = 0.0

    finally:
        server.stop()`

oroulet, 2017-03-08:
Instead of global. use a class

class Myclass():
    def __init__(self, generator):
          selg.generator = generator
    @uamethod
    def Stop(self, parent):
       whatever

thinking about it, maybe the decorator is not very happy about this. But you can test

zerox1212, 2017-03-08:
I agree. You should make a class.

Perhaps it is time that I publish some of my code as an example that makes a Python Object mirror a UA Object. It would be useful for this case. Then you only need to program one "object" instead of two.

destogl, 2017-03-15:
@aquari89 Look at the line 60 of server-events.py example

destogl, 2019-07-31:
could this be closed? Since is inactive for two years and we provided an answer?

closed 2019-08-02 by destogl