FreeOpcUa/python-opcua #418: History read - timeout

kapsl, 2017-03-21:
Hello together,
I really like the python-opcua client. But currently I have following problem:

We record data over a longer period of time, and it gets quite a bit of data (perhaps. Normally the history read works perfect, but at some point (I guess, when the amount of data has exceeded some threshold). I always get a timeout error for history read. 
I tried setting the timeout for the client to some large value, but this doesn't work. 

It is a strange behaviour. At the beginning the time to retrieve the history values is increasing (probably linear) but at some point it doesn't work anymore. I think the amount to transfer is around 8MB where it gets stuck.

Thanks,
lg Manu

oroulet, 2017-03-21:
ua defines several timeouts. Look at code in ua_client and search for timeout, I do not remember exactly how to setup them

kapsl, 2017-03-21:
I just find the timeout which is given in the Constructor Client(timeout=60). This value goes to the UaClient class. Other timeouts are very big, hours or days, so it can't have anything to do with it...

zerox1212, 2017-03-21:
It could be the server that gives up. I think there are also some timeouts in the UA spec for how long a server will process a request. Maybe you could look at the server parameters.

kapsl, 2017-03-21:
But when I access the history with UaExpert it works...

zerox1212, 2017-03-21:
In that case we will need to investigate.

oroulet, 2017-03-22:
OK. then fire up wireshark (google opcua and wireshark to find a tutorial) and record a session with uaexprt and one with python-opcua and try to spot the difference. A timeout parameter must be different somewhere.
You can attach the sessions to this bug report if you do not find out. And do not forget to let us now what you found.

kapsl, 2017-03-23:
Hi,
ok I managed to get the connection between prosys (server) and history read python-opcua in wireshark. At the moment I'm struggling to record the UaExpert traffic, because it runs on the lockal Loopback adapter...
[python_opcua_history_request.zip](https://github.com/FreeOpcUa/python-opcua/files/863947/python_opcua_history_request.zip)

oroulet, 2017-03-23:
The easiest thing to do is to use an external PC to run uaexpert

kapsl, 2017-03-28:
Ok here finally now the uaexpert dump is...
[uaExpertTraceOpcua.zip](https://github.com/FreeOpcUa/python-opcua/files/875796/uaExpertTraceOpcua.zip)

And here from python ocpua:

[python_opcua_history_request_opcua.zip](https://github.com/FreeOpcUa/python-opcua/files/875801/python_opcua_history_request_opcua.zip)