FreeOpcUa/python-opcua #419: example/server-with-encryption.py

aquari89, 2017-03-22:
Hello, All 

There was a problem running server-with-encryption.py.
The following error has occurred.
[image]

Do you know what the cause is?

oroulet, 2017-03-22:
you are missing some requirements (cryptography module I guess). look at README

aquari89, 2017-03-22:
I had some problems installing cryptography, but, after installed it, the server-with-encryption.py worked fine.
Thank you for your help.

closed 2017-04-18 by destogl

WangAllen, 2018-04-17:
Thanks for all, 
in running server-with-encryption.py, The following error has occurred.
what should i do next?

add_node: while adding node NumericNodeId(i=15957), requested parent node NumericNodeId(i=11715) does not exists
add_node: while adding node NumericNodeId(i=15958), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15959), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15960), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15961), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15962), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15963), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15964), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=16134), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=16135), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=16136), requested parent node NumericNodeId(i=15957) does not exists
Listening on 0.0.0.0:8080

zerox1212, 2018-04-17:
Those are probably just warnings when creating the standard address space. Your server is running and you should be able to connect to it with a client (that has a matching security cert).

WangAllen, 2018-04-17:
Thank you first
I am a new to opc, i am learning it.
when i am running client-minimal.py in example,it seems to be right,but no response showed in the server-minial,
when running client-example.py,there is the question in client,and the server is still no change,always"listening on 0.0.0.0:4840"
could you tell me what's wrong here,thanks
[image]
[image]

WangAllen, 2018-04-17:
Thank you first
I am a new to opc, i am learning it.
when i am running client-minimal.py in example,it seems to be right,but no response showed in the server-minial,
when running client-example.py,there is the question in client,and the server is still no change,always"listening on 0.0.0.0:4840"
could you tell me what's wrong here,thanks
/home/python/Desktop/2018-04-17 15-41-27屏幕截图.png
/home/python/Desktop/2018-04-17 15-42-25屏幕截图.png

zerox1212, 2018-04-17:
I don't understand what you are trying to do.

Your first screenshot looks correct. You have a running server on the left. On the right you have a client that connected to the server, browsed a few nodes, then disconnected.

Before you flood the github issues with questions you should read up on OPC UA itself. It's a complex protocol. You need a basic understanding of what a node is, how the address space is organized (name spaces and node IDs), and how subscriptions work for reading live values from a server.