FreeOpcUa/python-opcua #422: client error getting event notifications

mellotanica, 2017-03-27:
i'm writing a simple client that connects to a server and subscribes to Server node for BaseEventType events, i followed the lead of the client examples and created a small handler class with its `event_notification(self,event)` function and used an instance of this handler to receive notifications from a server.

every time e try to run the client, i get the next traceback:
```
Exception while calling user callback: %s
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/opcua/client/ua_client.py", line 473, in _call_publish_callback
    callback(response.Parameters)
  File "/usr/lib/python3.6/site-packages/opcua/common/subscription.py", line 113, in publish_callback
    self._call_event(notif)
  File "/usr/lib/python3.6/site-packages/opcua/common/subscription.py", line 150, in _call_event
    result = events.Event.from_event_fields(data.mfilter.SelectClauses, event.EventFields)
  File "/usr/lib/python3.6/site-packages/opcua/common/events.py", line 108, in from_event_fields
    ev.add_property(name, fields[idx].Value, fields[idx].VariantType)
IndexError: list index out of range
```

i did check with a debugger and during the execution of `from_event_fields` function, the fields list is actually empty, and it was like that from the beginning of the trace..

what should i do about this?

also i tried and interacted with the server doing something else than subscribing to events and everything works fine; the event seems fine for other opcua clients..

zerox1212, 2017-03-27:
Can you post your code?

oroulet, 2017-03-27:
This is strange a server is supposed to send back one field per field in subscription filter. You need to register a session with wireshark and attach the result to this bug request.

mellotanica, 2017-03-28:
ok, the code sits [here](https://pastebin.com/KiEBHwsT) and the trace of execution is attached below.

also, the complete output of the execution is the following:
```
connecting to opc.tcp://my.server.addr:port/..
connection ok, subscribing to event notifications..
Exception while calling user callback: %s
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/opcua/client/ua_client.py", line 473, in _call_publish_callback
    callback(response.Parameters)
  File "/usr/lib/python3.6/site-packages/opcua/common/subscription.py", line 113, in publish_callback
    self._call_event(notif)
  File "/usr/lib/python3.6/site-packages/opcua/common/subscription.py", line 150, in _call_event
    result = events.Event.from_event_fields(data.mfilter.SelectClauses, event.EventFields)
  File "/usr/lib/python3.6/site-packages/opcua/common/events.py", line 108, in from_event_fields
    ev.add_property(name, fields[idx].Value, fields[idx].VariantType)
IndexError: list index out of range
Exception while calling user callback: %s
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/opcua/client/ua_client.py", line 473, in _call_publish_callback
    callback(response.Parameters)
  File "/usr/lib/python3.6/site-packages/opcua/common/subscription.py", line 113, in publish_callback
    self._call_event(notif)
  File "/usr/lib/python3.6/site-packages/opcua/common/subscription.py", line 150, in _call_event
    result = events.Event.from_event_fields(data.mfilter.SelectClauses, event.EventFields)
  File "/usr/lib/python3.6/site-packages/opcua/common/events.py", line 108, in from_event_fields
    ev.add_property(name, fields[idx].Value, fields[idx].VariantType)
IndexError: list index out of range
```

[opcua_subscription_trace.zip](https://github.com/FreeOpcUa/python-opcua/files/874847/opcua_subscription_trace.zip)

oroulet, 2017-03-28:
so your server sends an empty event.... Did your other clients receive property values with the events? If so can you send a log of a session with them?

mellotanica, 2017-03-28:
sure, here is the log of a session with uaexpert client that receives events from the same server:

[opcua_subscription_working.zip](https://github.com/FreeOpcUa/python-opcua/files/875378/opcua_subscription_working.zip)

oroulet, 2017-03-28:
in your log they ask for 17 propertoes and the 17 properties are sent to client. This is normal. So we are probably doing something wrong or triggering a bug in server...but where.....

oroulet, 2017-03-30:
As a test I propose that you look into the code of events.py in get_filter_from_event_type function and let the whereclause uninitilized. I am guessing this will solve your issue. Stil I do not understand why the server sends us an empty event..

mellotanica, 2017-03-31:
> As a test I propose that you look into the code of events.py in get_filter_from_event_type function and let the whereclause uninitilized. I am guessing this will solve your issue. Stil I do not understand why the server sends us an empty event..

i can confirm that, if i comment out the line that assigns the whereclause the events are delivered correctly

zerox1212, 2017-03-31:
Is it possible for a server to not support the where clause?

mellotanica, 2017-03-31:
i'm doing some tests and it seems i receive a burst of events as long as there is a subscription active, much more than the server outputs..
my server is generating about one event per second but i get something like 150 events in a single seconds, each event has its unique time stamps and EventId, but they are a bit too much..

oroulet, 2017-03-31:
We do not make up these IDs, so the server must generate these events.

fkOnGit, 2017-10-05:
I have the same problem as described in [client error getting event notifications” #422](https://github.com/FreeOpcUa/python-opcua/issues/422):

When subscribing with a python client (based on client-events.py example) to the Server-node or other nodes of an OPC UA server, I get the following error message: 
`No handlers could be found for logger "opcua.client.ua_client"`.

Example code:
```python
import sys
sys.path.insert(0, "../python-opcua-0.90.3")
import time

from opcua import Client
from opcua import ua
from opcua import Subscription

class SubHandler(object):

    def event_notification(self, event):
        print("Python: New event recived: ", event)

if __name__ == "__main__":

    client = Client("opc.tcp://localhost:4880/", 60)  # TIMEOUT: 60s
    try:
        client.connect()
        handler = SubHandler()
        subscription = client.create_subscription(0, handler)

        # All events at "Server"-node:
        eventSubscription = subscription.subscribe_events()

        time.sleep(30) # wait for events

        subscription.unsubscribe(eventSubscription)
        #subscription.delete()

    finally:
        client.disconnect()
```

It seems that the server sends empty events (empty EventFields at ua.client.py at line 454, see the screenshot at the end of this comment), and the FreeOpcUa python client library has problems with that.

With a server that is based on the FreeOpcUa python-opcua sample server everything works fine. Maybe there is a different behavior between different OPC UA servers.

I’m using a sample server that is based on Softing’s OPC UA C++ Development Toolkit, Version 5.53, running on linux. (https://industrial.softing.com/en/products/software/opc-development-toolkits.html)

After subscribing to all events at the Server-node (or only subtypes of BaseEvent(Type) at other nodes) Softing’s Trace Viewer logs _“Content filter is not valid”_ from server.

Subscribing to the same events (as with the python client script) with the UaExpert client works without problems and all events are received correctly.

**A quick workaround is to modify the _events.py_ file:**
I disabled the where-clause in _events.py_ in _get_filter_from_event_type()_ as described in one of the other comments above.
When I now subscribe to all events at the Server-node I receive all events correctly. As well as subscribing to an own subtype of BaseEvent or only to events at a node below the Server-node. Then only the requested events are received.
But this is not a suitable solution for me, because I want to use the FreeOpcUa library as it is, without patching it.

I hope this explanation helps to find a solution for that issue. If further information are needed, I will try to collect these.

Here is a screenshot of the debugger where the empty fields are displayed:
[image]

oroulet, 2017-10-05:
Is the sample server available for us?

fkOnGit, 2017-10-09:
Hi @oroulet the SDK can be downloaded from [Softing C++ SDK](https://industrial.softing.com/en/products/software/opc-development-toolkits.html). There are some limitations (e.g. the server runs only for 90min without a licence key) but for reproducing this issue it is fine.

You can use the `<SDK_DIR>/Source/Apps/Samples/Cpp/Tutorial/Server_5` server from the SDK tutorials. It runs out of the box when you have setup the PKI.

If I can help you setting it up, feel free to contact me.

laimat, 2020-03-11:
Hi,
Today I ran into the same problems as described by @mellotanica by using the python async ua client with an open62541 based Server. 
Is there any fix or does anyone has an idear allready why this happens.
`Exception while calling user callback: %s
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/asyncua/client/ua_client.py", line 528, in _publish_loop
    callback(response.Parameters)
  File "/usr/local/lib/python3.7/dist-packages/asyncua/common/subscription.py", line 101, in publish_callback
    self._call_event(notif)
  File "/usr/local/lib/python3.7/dist-packages/asyncua/common/subscription.py", line 134, in _call_event
    result = Event.from_event_fields(data.mfilter.SelectClauses, event.EventFields)
  File "/usr/local/lib/python3.7/dist-packages/asyncua/common/events.py", line 113, in from_event_fields
    ev.add_property(name, fields[idx].Value, fields[idx].VariantType)
IndexError: list index out of range`

RobertHue, 2022-02-26:
Hey,

I ran into the same error:

There seems to be an **SimpleOverflowEventType** (id=4035) being send to the OPC UA client, which is fine, because it is **OfType** of **SomethingEventType** (id=1047).

See Image for the SimpleOverflowEventType:
[image]

After further investigation it turned out to be the following:
[image]

The problem with this is that the client does not receive any further events anymore after this event type got received. This event that got received only contains one event field; see:

[image]

As you can see there are two event_list entries. The first one being the id=4035 hence the **SimpleOverflowEventType**...
The program immediately exists here without processing the second event_list entry anymore...

Whereas, in the UAExpert client there is no such event being sent although I subscribed to every event type and its root node. 
In the UAExpert Client the properties of the events look as follows:

[image]

It may makes sense to take the same values for our code as well.
-> Increasing the **QueueSize** fixed it; see as follows:

[image]

I hope this helps if someone seems to be getting the **SimpleOverflowEventType** as well resulting in the **"list index out of range"**.

TommyTomson585, 2023-03-30:
Hello @ all.

Is there any further solution meanwhile? Or is it still the best way to disable the where-clause in line  get_filter_from_event_type()  in the events.py