FreeOpcUa/python-opcua #424: AnalogItemType with optional field EngineeringUnit cannot be parse from xml

XB32Z, 2017-03-29:
Hi, 

I have the following error
`File "C:\Python27\lib\site-packages\opcua\common\xmlimporter.py", line 234, in _set_attr
    for vtype, v2 in val:
ValueError: need more than 1 value to unpack`
when I am trying to import a xml such as:
```
<UAVariable DataType="EUInformation" ParentNodeId="ns=1;i=6076" NodeId="ns=1;i=6166" BrowseName="EngineeringUnits">
        <DisplayName>EngineeringUnits</DisplayName>
        <References>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=6076</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
        </References>
        <Value>
            <uax:ExtensionObject>
                <uax:TypeId>
                    <uax:Identifier>i=888</uax:Identifier>
                </uax:TypeId>
                <uax:Body>
                    <uax:EUInformation>
                        <uax:NamespaceUri>http://www.opcfoundation.org/UA/units/un/cefact</uax:NamespaceUri>
                        <uax:UnitId>5066068</uax:UnitId>
                        <uax:DisplayName>
                            <uax:Text>mm</uax:Text>
                        </uax:DisplayName>
                        <uax:Description>
                            <uax:Text>millimetre</uax:Text>
                        </uax:Description>
                    </uax:EUInformation>
                </uax:Body>
            </uax:ExtensionObject>
        </Value>
    </UAVariable>
```
This node is a property attached to a AnalogDataType variable.
Thx for your help

zerox1212, 2017-03-29:
I looked quickly and it requires debugging because it appears the importer is trying to make a list instead of an object.

XB32Z, 2017-03-30:
I added a pull request #425 to solve this issue. The val was instance of unicode not str
Thx for your support and for doing this :)

zerox1212, 2017-03-30:
I think you should try opening your XML with notepad++ and then save it as a UTF-8 encoding. Then try to import it.

closed 2017-03-31 by XB32Z