FreeOpcUa/python-opcua #426: Problems with custom enumeration from xml import

devel0peris, 2017-04-03:
Hey there.  I import the following XML into my server:
``` xml
...
    <UAObjectType BrowseName="0:CustomType" NodeId="i=20001">
        <DisplayName>CustomType</DisplayName>
        <Description>CustomType</Description>
        <References>
            <Reference IsForward="false" ReferenceType="HasSubtype">i=58</Reference>
            <Reference ReferenceType="HasProperty">i=20003</Reference>
            <Reference ReferenceType="HasProperty">i=20007</Reference>
        </References>
    </UAObjectType>
    <UADataType BrowseName="0:Prio" NodeId="i=20004">
        <DisplayName>Prio</DisplayName>
        <Description>Prio</Description>
        <References>
            <Reference IsForward="false" ReferenceType="HasSubtype">i=29</Reference>
            <Reference ReferenceType="HasProperty">i=20005</Reference>
        </References>
    </UADataType>
    <UAVariable ArrayDimensions="3" BrowseName="0:EnumStrings" DataType="LocalizedText" NodeId="i=20005" ParentNodeId="i=20004" ValueRank="-3">
        <DisplayName>EnumStrings</DisplayName>
        <Description>EnumStrings</Description>
        <References>
            <Reference IsForward="false" ReferenceType="HasProperty">i=20004</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
        </References>
        <Value>
            <uax:ListOfLocalizedText>
                <uax:LocalizedText>
                    <uax:Text>Prio1</uax:Text>
                    <uax:Locale/>
                </uax:LocalizedText>
                <uax:LocalizedText>
                    <uax:Text>Prio2</uax:Text>
                    <uax:Locale/>
                </uax:LocalizedText>
                <uax:LocalizedText>
                    <uax:Text>Prio3</uax:Text>
                    <uax:Locale/>
                </uax:LocalizedText>
            </uax:ListOfLocalizedText>
        </Value>
    </UAVariable>
    <UAVariable BrowseName="0:prio" DataType="i=20004" NodeId="i=20007" ParentNodeId="i=20001">
        <DisplayName>prio</DisplayName>
        <Description>prio</Description>
        <References>
            <Reference IsForward="false" ReferenceType="HasProperty">i=20001</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
        </References>
        <Value>
            <uax:LocalizedText>
                <uax:Text>Prio3</uax:Text>
                <uax:Locale/>
            </uax:LocalizedText>
        </Value>
    </UAVariable>
...
```

I add the python object with the following code:
``` python
    def add_object(self, name: str, objecttype: str, val=None):
        nodeid = self.server.get_root_node().get_child(["0:Types", "0:ObjectTypes", "0:BaseObjectType", "0:{}".format(objecttype)]).nodeid
        myobject = self.myobj.add_object(self.idx, name, nodeid)

        for attr in myobject.get_children():
            name = attr.get_display_name().to_string()
            attr.set_value(getattr(val, name))
```

Everything works fine, but when I try to read the custom enum value _prio_ the server crashes:
```
Exception raised while parsing message from client, closing
Traceback (most recent call last):
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\server\binary_server_asyncio.py", line 85, in _process_data
    ret = self.processor.process(hdr, buf)
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\server\uaprocessor.py", line 87, in process
    return self.process_message(msg.SecurityHeader(), msg.SequenceHeader(), msg.body())
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\server\uaprocessor.py", line 107, in process_message
    return self._process_message(typeid, requesthdr, algohdr, seqhdr, body)
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\server\uaprocessor.py", line 183, in _process_message
    self.send_response(requesthdr.RequestHandle, algohdr, seqhdr, response)
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\server\uaprocessor.py", line 41, in send_response
    response.to_binary(), message_type=msgtype, request_id=seqhdr.RequestId, algohdr=algohdr)
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\ua\uaprotocol_auto.py", line 8969, in to_binary
    packet.append(fieldname.to_binary())
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\ua\uatypes.py", line 988, in to_binary
    packet.append(self.Value.to_binary())
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\ua\uatypes.py", line 835, in to_binary
    b.append(uabin.pack_uatype(self.VariantType, self.Value))
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\ua\ua_binary.py", line 267, in pack_uatype
    return extensionobject_to_binary(value)
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\ua\uaprotocol_auto.py", line 16373, in extensionobject_to_binary
    TypeId = FourByteNodeId(getattr(ObjectIds, "{0}_Encoding_DefaultBinary".format(obj.__class__.__name__)))
AttributeError: type object 'ObjectIds' has no attribute 'Prio_Encoding_DefaultBinary'
```

Can someone please tell me the fault?

oroulet, 2017-04-03:
do you used a released version or master? I just rewote that part in master. Can you test?

I do not think it will fix it anyway. I am not sure what is happening here. Are you following the procesdure in examples/server-custom-object.py ?
mayne you are writting bad data using set_value(), we do not do any type check on server side...

devel0peris, 2017-04-03:
I used a released version. Using the master I got the following Traceback:
```
Exception raised while parsing message from client, closing
Traceback (most recent call last):
...
  File "...\python-opcua\opcua\ua\uatypes.py", line 1172, in extensionobject_to_binary
    TypeId = extension_object_ids[obj.__class__.__name__]
KeyError: 'Prio'
```

The method _set_value(value)_ is called with an enum _Prio.Prio3_.
Thank you for your support!

oroulet, 2017-04-03:
Enums in ua are not really enum, they are strings. Write a string there.

Maybe we should see if it is possible to raise a more understandable error for users

devel0peris, 2017-04-04:
Thank you very much! Now it works.
Maybe it would be helpful to document the usage of enums and other special datatypes.

bitkeeper, 2017-04-24:
Hi guys,
Little bit late for the discussion but I notice a few things:

1.
@oroulet Can you explain what you mean with are enum are strings ?

Enums are coded over the wire as numbers (see Standard Part 6 5.2.3). The set and get of an enum is by this always an Int32.  The EnumStrings are only used to make it more human to read/write, but are not required to get stuff working.
Due this I doubt if de common test test_enum implementation is correct; I don't expect get_value to return an localized string.

2.
@devel0peris Concerning custom types try keep away from namespace 0 in code. python-opcua treat stuff in namespace different.

In xml it is fine if you have a namespace declaration on top like, then on import al your definitions are remapped to the a new namespace index:
```
    <NamespaceUris>
        <Uri>http://www.yournamespace.nl/blabla</Uri>
    </NamespaceUris>
```

3.
@devel0peris If you apply the above to your enum value xml file the value should be changed to:
```xml
<Value>
      <uax:Int32>2</uax:Int32>
 </Value>
```

By off course by joining after the discussion it can happen that I totally on the wrong track ;-)

oroulet, 2017-04-24:
@bitkeeper I do not remember the details but there are two types of enum EnumStrings and EnumValues. one is a string, the other an extension object.  The second type might be instanciated as uint32..

bitkeeper, 2017-04-25:
@oroulet 
EnumStrings and EnumValues are both coded as Int32 (see Standard Part 6 5.2.3). Compare it with enums in C. Take a look at `uaprotocol_auto.py` it contains examples of both for example ApplicationType vs AttributeWriteMask.
Think about what it would mean to use LocalizedText as values for clients  ... 

EnumStrings and EnumValues are only used giving a specific meaning to Int32 value by using label. And has nothing todo with the value transport itself. 

The difference between EnumStrings and EnumValues is that EnumString is a fixed sequence and EnumValues allow gaps. 

EnumStrings Example:
- Label 'A' - Int32 value 0
- Label 'B' - Int32 value 1
- Label 'C' - Int32 value 2

EnumValues Example:
- Label 'A' - Int32 value 3
- Label 'B' - Int32 value 6
- Label 'C' - Int32 value 8

EnumStrings can just be described as a list because it a sequence (no gaps in the numbers).
That is why EnumValues are using a struct (extension object) to describe the relation between the Int32 number and the label.

Attached is a zip that contains a xml schema example for EnumStrings and EnumValues:
[ua_xmlschema_enumexample.zip](https://github.com/FreeOpcUa/python-opcua/files/954197/ua_xmlschema_enumexample.zip)

oroulet, 2017-04-25:
@bitkeeper Interresting. What you say make sense. Do we have en error in implementation? I kind of remember having to write string to enums, but maybe I remember wrong....

bitkeeper, 2017-04-25:
@oroulet ok did some enum testsing today, conclusie at the end:

**Created opc xml schema with:**
- custom EnumStrings type
- custom EnumValues type
- object instance with
  - var of custom EnumStrings type
  - var of custom EnumValues type

**Test 1 - load xml schema and inspect the server**  
- Did an xml import with the python-opcua server.
- Browsed the nodes with UExpert 
- Changed the values with a picklist

**Ok**: everything seems fine

**Test 2 - change value of enum at server side**  
- Did an xml import with the python-opcua server.
- did an evar.set_value(0) in the code
- Browsed the nodes with UExpert

**Nok**: No label more present with the value
- Used a dataview to inspect the value

Expected wrong type due no type check at the server side.
Indeed wrong type: Int64 
- did an evar.set_value(ua.Variant(1, ua.VariantType.Int32))

**Ok**: fine now type ok and label keeps present.

**Test 3 - Use value based on IntEnum (like in uaprotocol_auto.py)**
- Create enum based on IntEnum
- Did an xml import with the python-opcua server.
- Did an var.set_value with the IntEnum based version

**Nok**: The IntEnum also maps to Int64 instead of Int32

**Test 4 - browse existing type I**
- Start python-opcua server
- Browse with UAExpert to BaseDataTypes > Enumeration > AttributeWriteMask
- Inspect EnumValues

**Ok**

**Test 5 - browse existing type II**
- Start python-opcua server
- Browse with UAExpert to BaseDataTypes > Enumeration > AxisScaleEnumeration
- Inspect EnumStrings

**Nok**: raise UaError("{0} could not be packed with value {1}".format(vtype, value))

The part that isn't going well is:
`  standard_address_space_part8.py:953   attrs.Value = ua.Variant(['Linear', 'Log', 'Ln'], ua.VariantType.LocalizedText)`
  
Reproducable by:
```
v=ua.Variant(['Linear', 'Log', 'Ln'], ua.VariantType.LocalizedText)
v.to_binary()
```

or even more simple:
```
v=ua.Variant('Linear', ua.VariantType.LocalizedText)
v.to_binary()
```

**Preliminary Conclusion:**
- Load custom enums from xml works just fine
- Just use Int32 when setting Enum Values
- The  enum generated in uaprotocol_auto.py based on  IntEnum have in the end the wrong data type in ua (Int64 instead Int32)
- to_binary for ua.Variant(['Linear', 'Log', 'Ln'], ua.VariantType.LocalizedText) and ua.Variant('Linear', ua.VariantType.LocalizedText) fails

bitkeeper, 2017-04-25:
@oroulet 
I also have doubts about the implementation `ua_utils.py:data_type_to_variant_type` regarding a  enumeration.
If now returns `ua.VariantType.LocalizedText` or `ua.VariantType.ExtensionObject`. I think it should be `ua.VariantType.UInt32` ?

oroulet, 2017-04-25:
I got a crash there on an enum last week so you are probably right

oroulet, 2017-04-25:
If you have 2 minutes, the best would be to write a small test showing the
error

bitkeeper, 2017-04-25:
Haha decent tests in 2 minutes ... I will see if I find some time tomorrow.

oroulet, 2017-04-25:
@bitkeeper I went to all you comment above and I could not see any error.
* When calling set_value we do not check the type of node, and default for numbers is in64, so no bug there. Adding check for data type would be very expensive in python
* v=ua.Variant('Linear', ua.VariantType.LocalizedText) is wrong. some time ago we had a test in Variant construtor to generate on-the-fly a LocalizedText object but it was removed.
So basically we are down to fundamental type checking issues that I am not sure we can solve in python in an efficient way

bitkeeper, 2017-04-25:
@oroulet 
I don't have a problem with set_value at all, it was only a warning that with enums you shouldn't set_value with the default numbers, but explicit use use a Int32 else you end up with Int64. 

I think we have 3 things that requires fixing:
1. ua_utils.py:data_type_to_variant_type (already fixed see PR #439 )
2. Using generated classes in uaprotocol_auto.py based on IntEnum as values result in Int64 instead of Int32.
3. All the construction like ua.Variant(['Linear', ...], ua.VariantType.LocalizedText) in standard_address_space_part8 will give an exception when a client tries to get the enum labels.
You can say it is wrong, but it is the generated code in the standard adress space that uses this construction. If `ua.Variant('Linear', ua.VariantType.LocalizedText)` is wrong what is correct ?Can we just replace it by ua.LocalizedText('Linear')?
In that case the code generator should be updated.

oroulet, 2017-04-25:
2. Ok we might have a bug here.
3. Yes