FreeOpcUa/python-opcua #427: About the arguments of add_object ...

aquari89, 2017-04-05:
As I know, The arguments of add_object are:
```
  def add_object(self, nodeid, bname, objecttype=None):
        return opcua.common.manage_nodes.create_object(self, nodeid, bname, objecttype)
```

However, In example/server-custom-object.py, I saw that there is a **nodeid** in the add_object argument.
```
    # get nodeid of custom object type by one of the following 3 ways:
    # 1) Use node ID
    # 2) Or Full path
    # 3) Or As child from parent
    myobject1_type_nodeid = ua.NodeId.from_string('ns=%d;i=2' % idx)
    myobject2_type_nodeid = server.get_root_node().get_child(["0:Types", "0:ObjectTypes", "0:BaseObjectType", "%d:MyCustomObjectType" % idx]).nodeid
    myobject3_type_nodeid = server.get_node(ua.ObjectIds.BaseObjectType).get_child(["%d:MyCustomObjectType" % idx]).nodeid

    # populating our address space    
    myobj = objects.add_object(idx, "MyCustomObjectB", myobject3_type_nodeid)
```

zerox1212, 2017-04-05:
Are you talking about `idx` ?

aquari89, 2017-04-06:
No. I'm talking about **objecttype**.

Can I use `nodeid `instead of `objecttype `in the argument of add_object ?

oroulet, 2017-04-06:
objecttype should refer to the node representing the type. it is probably possible to write a nodeid, a Node or a numeric value (then namespace 0 is assume)
The default is probably the nodeid of BaseObjectType