FreeOpcUa/python-opcua #428: xml import - node manage

aquari89, 2017-04-07:
For example, import a new object from xml address space and create a instance of the new object type.

The xml('customobject.xml') address space is as follows:
```
        + ControllerType(Custom Object Type)
   		+ machineTemperature(Variable) - Temperature estimated Value
   		+ TemperatureSetPoint(variable) - Temperature_standardPoint
   		+ StartWithSetPoint(method) - Temperature_standardPoint
                        +inputArgument
```
To manage variable node, I programmed the server as follows.
Is it right to use get_child to manage variable node?

The server is as follows:
```
    if __name__ == "__main__":

    server = Server()
    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")

    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)

    objects = server.get_objects_node()

    server.import_xml('customobject.xml')

    obj_type_nodeid = server.get_node(ua.ObjectIds.BaseObjectType).get_child(["%d:ControllerType" % idx]).nodeid

    myobj = objects.add_object(idx, "MyCustomObjectB", obj_type_nodeid)
    machineTemperature = myobj.get_child(["%d:machineTemperature" % idx])

    server.start()
    try:
        while True:
            time.sleep(1)

            temperature = random.uniform(-3.0, 3.0)
            machineTemperature.set_value(temperature, varianttype = ua.VariantType.Double)
            temperature = 0.0

    finally:
        server.stop()
    ...
```

Plus One more question, How can I instantiate and invoke a method when using xml address space?

When I created the custom type directly in the source code, I was able to call the method.
However, I do not know how to handle the method when importing xml.

oroulet, 2017-04-07:
there many way to get a proxy (Node object) to a node. But get_child is often the most human friendly if you know the path or get_node(nodeid) if you know the nodeid.
To call a method defined in xml you need to link that method to a python method or function (look at examples), get a proxy/Node to parent node and call method with call_method method from Node

zerox1212, 2017-04-07:
There currently is not an example. Can read here https://github.com/FreeOpcUa/python-opcua/issues/297
 or look in server.py:

```python
def link_method(self, node, callback):
        """
        Link a python function to a UA method in the address space; required when a UA method has been imported
        to the address space via XML; the python executable must be linked manually
        Args:
            node: UA method node
            callback: python function that the UA method will call
        Returns:
        """
        self.iserver.isession.add_method_callback(node.nodeid, callback)
```

zerox1212, 2017-04-07:
@oroulet Why don't these doc strings show up in the documentation web page?

oroulet, 2017-04-07:
They should....I do not remember how this works..

zerox1212, 2017-04-07:
Sorry, I found it. It does work, but it sorts it differently than it appears in the code. Disregard that message.