FreeOpcUa/python-opcua #429: Creating a Cache File for faster loading on Raspberry Pi (Server)

bullvai, 2017-04-08:
[image]

Please forgive my ignorance, but how is this achieved? Does anyone have a specific example of how they've implemented this? My Raspberry Pi 3b never had any issues but the Pi Zero W suffers from very slow loading. Any help would be greatly appreciated!

zerox1212, 2017-04-10:
When you instantiate a server just do this:
```python
myserver = server(shelffile="cache")
```
You can read the doc strings here as well: https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/server/address_space.py#L515

bullvai, 2017-04-11:
I end up with an error - NodeId object is not iterable:
[image]

zerox1212, 2017-04-11:
That was fixed here: https://github.com/FreeOpcUa/python-opcua/pull/421 but it isn't in the released package. Take the code from the master.

oroulet, 2017-04-12:
There is a large change for extension objects in master which I was afraid to release. but maybe we should just release it....

bullvai, 2017-04-12:
I edited the code and changed .keys to .items. After doing so I changed the following in my python3 script:

Instead of server = Server() I wrote:
server = Server(shelffile="cache")

In the directory my script is located in, a file named "cache.db" was created, and I received no complaints from python. However, the server still took 3 minutes to initiate, so I feel as though something is still wrong.

zerox1212, 2017-04-13:
The first start up will take a long time. After that start up should be faster because the lazy loading dict is used.

tennfy, 2017-07-28:
I also have this problem.

I use server = Server(shelffile="cache") to initiate, the first time it creates  cache.dir & cache.pag file. But the second time it still took about 5 minitues.

 @bullvai Do you solve this problem?

KingUdo, 2017-07-29:
I have the same issue, but in the Model B+. The boot up times are really long, and I can't get it fixed by using the server = Server(shelffile="cache") trick.
Any help appreciated ;-)

zerox1212, 2017-07-30:
Are you using the release from pip, or the latest master? Honestly I have not tested the caching in a long time. The best thing to do would be to debug your server when you start it and follow `shelffile=` through the many constructors as your server is starting up and make sure it's really using the python shelf.

bullvai, 2017-07-31:
Sorry for the delayed response. I never ended up solving the problem; I just accepted that the server will have a slow load time. =/

oroulet, 2017-07-31:
just looked at the code and it was obviously broken. I just pushed a commit to master. Can someone test it? The second start should be must faster

tennfy, 2017-08-01:
@oroulet Great work

bullvai, 2017-08-07:
I'm still having trouble with this. @zerox1212 you mentioned the LazyLoadingDict class. How do I implement this?

zerox1212, 2017-08-07:
Set a break point here https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/server/internal_server.py#L90 and then step through the code when you server starts up to make sure it's loading the address space from the shelf.

oroulet, 2017-08-07:
@zerox1212 👍

zerox1212, 2017-08-07:
@bullvai According to your profile your an automation guy, if you need help with object oriented programming let me know and I can probably find some way to show you how to trouble shoot python.

bullvai, 2017-08-08:
That would be awesome @zerox1212! I program PLCs in SCL (Structured Text), which is similar-ish to Pascal, and is not object oriented.

bullvai, 2017-08-08:
Should I be using the pdb module to set these breakpoints? Also, I'd like to talk more about the memory leak issue - should I make a new issue thread or just put it here?

oroulet, 2017-08-08:
Make a new issue for memory leak. And print is probably more than enough to
debug the very simple code of shelve

zerox1212, 2017-08-08:
Post an email address and I can email you to help you. Then we don't clutter up this issue.

bullvai, 2017-08-09:
@oroulet I've been doing some testing and it seems that the memory leak issue is gone now. I could post my new results vs my old results in another thread if you're still interested, but it doesn't seem to be an issue anymore so I'm not certain if there needs to be an open issue for it.

bullvai, 2017-08-10:
@zerox1212 I sent you an email - let me know if you got it.

farkau, 2018-07-20:
Hi,
thanks for this nice tool :)

Loading the shelffile also did not work for me. The shelffile was always created and never loaded.
That might be why the shelffile seems to have no impact for some commentators here.

The reason seems to be python3 related:

[https://stackoverflow.com/questions/27493733/use-python-2-shelf-in-python-3](url)

**Short:** 
dbm in python3 generates shelffile+".db" while writing to the shelffile. 
dbm in python2 does not do that.
Server [https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/server/internal_server.py#L90](url) checks for shelffile without ".db". As a result it does not detect the created shelffile (tested with Python 2.7.1 and 3.5.2).

**Solution:**

`if shelffile is not None and (os.path.isfile(shelffile) or os.path.isfile(shelffile+".db")):`

I do not know which python versions are (and will be) affected, thus I prefer the simple additional condition as above.

What do you think?

zerox1212, 2018-07-20:
This looks correct. Unfortunately `shelffile` is not very good in cross platform situation. Windows, Linux, etc all make a different file name.