FreeOpcUa/python-opcua #431: Filtering / Searching for nodes

jvibo, 2017-04-12:
Hello all, 
I’m looking for a possibility to filter nodes by an expression. E.g. a list with all nodes that are Integers. I think there is some feature with events and where conditions, but I need to filter all nodes not only the event nodes.

oroulet, 2017-04-12:
On server side you can go through the entire addresspace dict. On client side I do not think ua defines such a possibility. If anybody knows one, let me know.

jvibo, 2017-04-12:
How do you achieve this? And is it implemented in python opc-ua yet?

oroulet, 2017-04-12:
you can filter events yes. you will need to look at code of subscribe_events and look how the filter is generated. everything should be implemented but I do not remember exactly what is possible and not