FreeOpcUa/python-opcua #435: Connection pythonopcua-Client & Unified Automation Server

aquari89, 2017-04-17:
Hello, All.
Currently I am creating a client using PythonOpcUa and testing the connection through the unified automation .Net example server.
However, an error occurs during the test.
[image]

I think this part (aircon1 = controlFold.get_child (["2: AirConditioner1"])) seems to be a problem.
I think I did exactly, but I do not know what the problem is.
If I have any mistakes, please let me know.

Client Code
```
import sys
sys.path.insert(0, "..")
import time

try:
    from IPython import embed
except ImportError:
    import code

    def embed():
        vars = globals()
        vars.update(locals())
        shell = code.InteractiveConsole(vars)
        shell.interact()

from opcua import Client
from opcua import ua

class SubHandler(object):

    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)

    def event_notification(self, event):
        print("Python: New event", event)

if __name__ == "__main__":

    client = Client("opc.tcp://10.0.0.171:48030")

    try:
        client.connect()

        root = client.get_root_node()

        objects = root.get_child(["0:Objects"])

        controlFold = objects.get_child(["2:Controllers"])

        aircon1 = controlFold.get_child(["2:AirConditioner1"])

        humidityVar = aircon1Obj.get_child(["2:Humidity"])
        humiditysetpointVar = aircon1Obj.get_child(["2:HumiditySetPoint"])
        stateVar = aircon1Obj.get_child(["2:State"])
        temperatureVar = aircon1Obj.get_child(["2:Temperature"])
        temperaturesetpointVar = aircon1Obj.get_child(["2:TemperatureSetPoint"])

        customEvent = root.get_child(["0:Types", "0:EventTypes", "0:BaseEventType", "3:ControllerEventType"])
		
        handler = SubHandler()
        sub = client.create_subscription(500, handler)
        handle = sub.subscribe_events(aircon1Obj, customEvent)
        time.sleep(0.1)

        sub.subscribe_events()

        res = machine_Obj.call_method("2:Start")
        print("method result is: ", res)

        embed()
        sub.unsubscribe(handle)
        sub.delete()

    finally:
        client.disconnect()
```

.Net example server.
[image]

http://documentation.unified-automation.com/uasdkdotnet/2.5.4/html/L3ServerTutGSLess06.html

oroulet, 2017-04-17:
A path call used browse name not node ID. The namespace of browse name is 3 here. Not 2

zerox1212, 2017-04-17:
You need `["3: AirConditioner1"]` according to your screenshot. I never did understand the reasoning behind NS and browse name index not matching...

oroulet, 2017-04-17:
My personal opinion is that this plainly stupide and probably the result of compatibility with an archaic system

aquari89, 2017-04-18:
Thanks!
I solved this problem.

JesseJ12345, 2019-04-30:
@aquari89 , Did you fix **Broken pipe**?  I also meet this problem