FreeOpcUa/python-opcua #437: Client History Read

aquari89, 2017-04-24:
Hello, All.

The Pythonopcua Client has confirmed that there is a function to read the history data.

`def history_read (self, params):`
reference : http://python-opcua.readthedocs.io/en/latest/_modules/opcua/client/ua_client.html?highlight=history%20read

If client use this function, can it read about events or data value changes?
I don't know this part because there is no explanation.

Currently I have implemented a test server with pythonopcua to get the history data (event or data value change) from the client.
However, I do not know what to do with the client to get this history data.

zerox1212, 2017-04-24:
That is the low level part of the component. You have to build all the required UA objects in params and send that request to the server.

You need to make a HistoryReadParameters object (https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/ua/uaprotocol_auto.py#L9570) with all data on the history stuff you want to read from the server. This isn't documented here, you would need to go on the OPC Foundation website and download the part of the spec that details reading history from a server. You can get a PDF of the spec, but you have to create an account.

aquari89, 2017-04-24:
@zerox1212 
Does this( "There is not a high level function to read the history.") mean that the history read function is not yet supported by the pythonopcua client?

In other words, do I have to implement it myself in order to use the history read function on the client?

oroulet, 2017-04-24:
Look at tests https://github.com/FreeOpcUa/python-opcua/blob/master/tests/tests_history.py#L167 . reading of event history is easy and fully supported. What is more complicated is if you want to filters server side the events you want to read. Then you will need to make a filter object

zerox1212, 2017-04-24:
I don't think that test is using client methods is it?

aquari89, 2017-04-25:
@oroulet 
What you say(tests https://github.com/FreeOpcUa/python-opcua/blob/master/tests/tests_history.py#L167 . ) is like talking on the server side.

I want to read history data from the client.

zerox1212, 2017-04-25:
It's hard for everyone to read lol. Best way is to just try it. Like I said before make a params object and just try to fill in all the fields; note that some fields might require other instantiated UA objects

```python

myparams = ua.HistoryReadParams()

myparams.x = ???
myparams.someotherattr = ???

myhistory = myclient.history_read(myparams)
```

oroulet, 2017-04-25:
the Node object has a method called history_read_events. Is this broken on client side? The code looks to be server/client independent and should work on both side...

oroulet, 2017-04-25:
I remember I wanted to make history test code run both on client and server. But I do not remember if I did it

oroulet, 2017-04-25:
looking at code, the tests are only run on server side. This is different that all other tests in python -opcua. Should fix it but I reall have not time currently

zerox1212, 2017-04-25:
@aquari89 
Can you try using the methods listed here: https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/common/node.py#L472 ?

aquari89, 2017-04-27:
@zerox1212 
I used read_event_history, and was able to read the history data(event).

Attach the following figure to show the result.
The first picture shows that a total of four events occurred.
[image]
The following figure shows the result of reading the history data (event) by implementing the test client using function(read_event_history).
[image]

luchs3, 2017-10-18:
Hello,
Is it already possible to read history data (values)? I'd need this option and better ask before I dig into the ua specifications.

zerox1212, 2017-10-19:
If the node has history enabled on the server side you can read it.

luchs3, 2017-10-19:
Ok great, but how do I set the starttime  and endtime?

zerox1212, 2017-10-19:
When you build your history request you set the start and end time. Look at https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/common/node.py#L460

luchs3, 2017-10-19:
Perfect! Thank you very much for your effort!

Midhunpottammal, 2023-07-12:
FreeOPCUA supports historical data access ?