FreeOpcUa/python-opcua #440: Code generator broken ?

bitkeeper, 2017-04-26:
Today I wanted to make some changes to the codegenerator output, but the generator doesn't work.

Because the generator isn't part of the current tests; Can it be that the code generator (schemas/generate_address_space.py:CodeGenerator) is broken due changes to the xmlparser last year ?

oroulet, 2017-04-26:
Can be...  Should not be hard to fix anyway

oroulet, 2017-04-26:
just checked and yes it is broken. I kind of remember that I had to change that to a list to preserve order. should be same data so easy to fix. hopefully....

bitkeeper, 2017-04-26:
Was indeed easy to fix. Because I not really familar with this piece of code can you review PR #442 and check if the output of the generator is still what you expect ?

zerox1212, 2017-04-26:
Would be great if you could add a few doc strings on how the code generator works. I tried to use it a long time ago to generate part 14 of specification and couldn't figure out how to get it to run...

PR #442 (2017-04-26): Fixed codegenerator due changes in the xmlparser
resolve #440
closed 2017-04-26 by oroulet