FreeOpcUa/python-opcua #443: save_node_value accuracy in sql history

brubbel, 2017-04-28:
Hi,
In save_node_value() ([linked here](https://github.com/FreeOpcUa/python-opcua/blob/848e3b964422759bc243dcc128db44a0434f632c/opcua/server/history_sql.py#L67)), the value of the data written to SQL is converted to a string as follows:
`str(datavalue.Value.Value)`
However, this results in a reduced accuracy for floats (although this value is only for human reading) e.g.
```
>>> import time
>>> a = time.time()
>>> str(a)
'1493390413.37'
```
The correct way would be:
```
>>> repr(a)
'1493390413.372789'
```

zerox1212, 2017-04-28:
You can make a patch if you want, but this field is only for looking at the database from another tool. UA server will always restore the binary from SQL which does not lose accuracy.

zerox1212, 2017-04-28:
Also if you are working on SQLite history feel free to clean up the code. There is some duplication that could be done better. It would make Scrutinizer happy. :)

brubbel, 2017-04-28:
Ok, I'll test it for some period and submit a patch.