FreeOpcUa/python-opcua #448: Intro Freeopcua

aquari89, 2017-05-11:
Hello, @oroulet .

It seems like a useless question, but I'm curious about the motivation for you to develop freeopcua.

Plus, Is FreeOpcUa a personal project? Or is FreeOpcUa working with a specific company?
If FreeOpcUa are working with a specific company, I would like to know about that company.

oroulet, 2017-05-11:
Well, I started contributed to the c++ stack, then decided to make a pure
python implementation to avoid compiling everywhere. Most of the work done
at work http://www.sintef.no/manufacturing where we make prototypes/ proof
of concepts for many different things and glue things together using Python

aquari89, 2017-05-12:
@oroulet 
Thank you for your kind answer.
I do not understand what this word(**to avoid compiling everywhere**) means in your answer. 
I would appreciate it if you could give me more details about **to avoid compiling everywhere**.

zerox1212, 2017-06-21:
He means that Python runs on Windows, Linux, and Mac without issues. C++ would have to be compiled for each OS. Having to compile often makes prototyping slow and less effective.