FreeOpcUa/python-opcua #449: Kepware connectivity: Exception: The user identity token is not valid.(BadIdentityTokenInvalid)

iammarco, 2017-05-11:
Hi Guys, I want to replicate your Kepware example 'client_to_kepware.py' but run into an authentication issue.
I use Python 3.5 on windows and installed the dependencies cryptography, dateutil, lxml and pytz.
I got Kepware KEPServerEX 6.0 V6.0.2107.0 installed on my local machine. I am using the base installation that is only the simualtion tags are available and I want to write a client reading simulation tag values.
In the OPC UA configuration I enabled opc.tcp://127.0.0.1:49320 and set secutiry to None. I also removed the flag from Basic128Rsa15 and Basic256.
My client is setip as follows:
`client = Client("opc.tcp://localhost:49320")` That is the Endpoint OPC U Configuration shows me
When executing connect I get below error. Can anybdoy help me on this? 

Thank u for any hints!

> 
WARNING:opcua.binary_client:ServiceFault from server received  in response to ActivateSessionRequest
Traceback (most recent call last):
  File "C:\Users\abcd\Python35-32\opcua.py", line 14, in <module>
    client.connect()
  File "C:\Users\abcd\Python35-32\opcua\client.py", line 102, in connect
    self.activate_session()
  File "C:\Users\abcd\Python35-32\opcua\client.py", line 180, in activate_session
    return self.bclient.activate_session(params)
  File "C:\Users\abcd\Python35-32\opcua\binary_client.py", line 221, in activate_session
    data = self._send_request(request)
  File "C:\Users\abcd\Python35-32\opcua\binary_client.py", line 61, in _send_request
    self._check_answer(data, " in response to " + request.__class__.__name__)
  File "C:\Users\abcd\Python35-32\opcua\binary_client.py", line 70, in _check_answer
    hdr.ServiceResult.check()
  File "C:\Users\abcd\Python35-32\opcua\uatypes.py", line 215, in check
    raise Exception("{}({})".format(self.doc, self.name))
Exception: The user identity token is not valid.(BadIdentityTokenInvalid)

aquari89, 2017-05-11:
Use the real address instead of the local host.
ex) `client = Client("opc.tcp://xxx.xxx.xxx.xxx:49320")`

iammarco, 2017-05-11:
I also tried 127.0.0.1 what is the value that is given but that did not change anything...tried my computers name in the network, too...

oroulet, 2017-05-11:
Are you missing username or password?

iammarco, 2017-05-11:
well, here i am unsure...as soon as i get's to authorization there is a gap in my knowledge. 
In Kepware i did not steup a user...not sure if that is even possible???I tried my windows user but i think that is not waht should be used...right? what user is required here?

oroulet, 2017-05-11:
No idea, it really depends on how your server is setup. But you can use one
of the the command lines to get supported endpoints. That should give you
the answer

iammarco, 2017-05-11:
Serversetup is simple...bascially I got Demo version from kepware page, installed by clicking next next next...Then i deactivated security in OPC UA settings. That's it.
I meanwhile also found out there is an option in Kepware Settings for creating users. I did so but still not working. I also tried all endpoint configurations Kepserver is offering...same result...what a pitty...

oroulet, 2017-05-11:
Several people have​ been using that server do it should work..

zerox1212, 2017-05-11:
Do you have another client that can successfully connect like UaExpert?

iammarco, 2017-05-11:
I tried with a comercial one - SAP PCo. Here it is working. Connection can be builld

iammarco, 2017-05-12:
folks, got myself a non-company laptop and rebuild scenario...now it is working. appears security is an issue here...still admin told me "on localhost there should not be any issue..."oh boy, a lot to learn here ;-)

oroulet, 2017-05-12:
Do not use Windows ;-)

iammarco, 2017-05-12:
dont tell me... ;-) thanks to all for providing help