FreeOpcUa/python-opcua #453: Client not terminating on TimeoutError

haru-a8n, 2017-05-27:
I am trying to stress test an OPC UA server pushing both CPU and memory usage. One of the error I am seeing which I think is expected is TimeoutError. The problem is that the script is not terminating. It seems like there is a thread stuck somewhere.

The is error I am expecting but the script is not closing:

2017-05-26 16:58:01,793|MainThread|C:\ProgramData\Miniconda3\lib\site-packages\opcua\client\ua_client.py|close_session|255|INFO|close_session
**********
Traceback (most recent call last):
  File "opcclientstress.py", line 55, in <module>
    client.disconnect()
  File "C:\ProgramData\Miniconda3\lib\site-packages\opcua\client\client.py", line 254, in disconnect
    self.close_session()
  File "C:\ProgramData\Miniconda3\lib\site-packages\opcua\client\client.py", line 471, in close_session
    return self.uaclient.close_session(True)
  File "C:\ProgramData\Miniconda3\lib\site-packages\opcua\client\ua_client.py", line 258, in close_session
    data = self._uasocket.send_request(request)
  File "C:\ProgramData\Miniconda3\lib\site-packages\opcua\client\ua_client.py", line 76, in send_request
    data = future.result(self.timeout)
  File "C:\ProgramData\Miniconda3\lib\concurrent\futures\_base.py", line 407, in result
    raise TimeoutError()

haru-a8n, 2017-05-27:
Additional log below:

Disconnecting
2017-05-25 16:08:26,413|MainThread|C:\ProgramData\Miniconda3\lib\site-packages\opcua\client\ua_client.py|close_session|255|INFO|close_session
**********
Traceback (most recent call last):
  File "opcclientstress.py", line 55, in <module>
    client.disconnect()
  File "C:\ProgramData\Miniconda3\lib\site-packages\opcua\client\client.py", line 254, in disconnect
    self.close_session()
  File "C:\ProgramData\Miniconda3\lib\site-packages\opcua\client\client.py", line 471, in close_session
    return self.uaclient.close_session(True)
  File "C:\ProgramData\Miniconda3\lib\site-packages\opcua\client\ua_client.py", line 258, in close_session
    data = self._uasocket.send_request(request)
  File "C:\ProgramData\Miniconda3\lib\site-packages\opcua\client\ua_client.py", line 76, in send_request
    data = future.result(self.timeout)
  File "C:\ProgramData\Miniconda3\lib\concurrent\futures\_base.py", line 407, in result
    raise TimeoutError()
concurrent.futures._base.TimeoutError
None
breaking out of while loop
closing keep alive session
2017-05-25 16:08:30,569|Thread-92826|C:\ProgramData\Miniconda3\lib\site-packages\opcua\ua\uaprotocol_hand.py|receive_from_socket|667|INFO|received header: Header(type:b'MSG', chunk_type:b'F', body_size:40, channel:1860881)

Observe "receive_from_socket" is fired after TimeoutError. Could this be a hint of the deadlock?

haru-a8n, 2017-05-27:
Below are some details for the threads. I think thread 7148 is the main thread, while 2980 is not. 

TID	CPU	Cycles Delta	Start Address	
7148			python.exe+0x131c	
2980			ucrtbase.dll!o__realloc_base+0x20	

===
Details of 2980 thread below
===

ntoskrnl.exe!KeSynchronizeExecution+0x3f26
ntoskrnl.exe!KeWaitForMultipleObjects+0x109c
ntoskrnl.exe!KeWaitForMultipleObjects+0xb3f
ntoskrnl.exe!KeWaitForSingleObject+0x377
ntoskrnl.exe!KiCheckForKernelApcDelivery+0x650
ntoskrnl.exe!KiCheckForKernelApcDelivery+0x27a
ntoskrnl.exe!KeWaitForMultipleObjects+0x1283
ntoskrnl.exe!KeWaitForMultipleObjects+0xb3f
ntoskrnl.exe!KeWaitForSingleObject+0x377
ntoskrnl.exe!NtWaitForSingleObject+0xf8
ntoskrnl.exe!setjmpex+0x3aa3
ntdll.dll!ZwWaitForSingleObject+0x14
mswsock.dll!Tcpip6_WSHOpenSocket2+0x2b3
mswsock.dll!Tcpip4_WSHGetWildcardSockaddr+0x1262
WS2_32.dll!recv+0x16c
_socket.pyd+0x31b0
_socket.pyd+0x1717
_socket.pyd+0x32c7
python36.dll!PyCFunction_FastCallDict+0x1d7
python36.dll!PyCFunction_FastCallKeywords+0x52
python36.dll!PyEval_GetFuncDesc+0x17a
python36.dll!PyEval_EvalFrameDefault+0x1101
python36.dll!PyEval_GetFuncDesc+0x2d2
python36.dll!PyEval_GetFuncDesc+0x36c
python36.dll!PyEval_GetFuncDesc+0x1e2
python36.dll!PyEval_EvalFrameDefault+0x1101
python36.dll!PyEval_GetFuncDesc+0x2d2
python36.dll!PyEval_GetFuncDesc+0x36c
python36.dll!PyEval_GetFuncDesc+0x1e2
python36.dll!PyEval_EvalFrameDefault+0x1101
python36.dll!PyEval_GetFuncDesc+0x2d2
python36.dll!PyEval_GetFuncDesc+0x36c
python36.dll!PyEval_GetFuncDesc+0x1e2
python36.dll!PyEval_EvalFrameDefault+0x1101
python36.dll!PyEval_GetFuncDesc+0x2d2
python36.dll!PyEval_GetFuncDesc+0x36c
python36.dll!PyEval_GetFuncDesc+0x1e2
python36.dll!PyEval_EvalFrameDefault+0x1101
python36.dll!PyEval_GetFuncDesc+0x2d2
python36.dll!PyObject_FastCallDict+0x6e
python36.dll!PyObject_Call_Prepend+0x94
python36.dll!PyObject_Call+0x78
python36.dll!PyEval_EvalFrameDefault+0x37b5
python36.dll!PyEval_GetFuncDesc+0x2d2
python36.dll!PyEval_GetFuncDesc+0x36c
python36.dll!PyEval_GetFuncDesc+0x1e2
python36.dll!PyEval_EvalFrameDefault+0x1101
python36.dll!PyEval_GetFuncDesc+0x2d2
python36.dll!PyEval_GetFuncDesc+0x36c
python36.dll!PyEval_GetFuncDesc+0x1e2
python36.dll!PyEval_EvalFrameDefault+0x1101
python36.dll!PyEval_GetFuncDesc+0x2d2
python36.dll!PyObject_FastCallDict+0x6e
python36.dll!PyObject_Call_Prepend+0x94
python36.dll!PyObject_Call+0x78
python36.dll!PyOS_SigintEvent+0x1ce3
python36.dll!PyThread_init_thread+0x4b2
ucrtbase.dll!o__realloc_base+0x60
KERNEL32.DLL!BaseThreadInitThunk+0x14
ntdll.dll!RtlUserThreadStart+0x21

oroulet, 2017-05-29:
If this is completely solve. please close the issue, And Thank again for looking at that hard to debug case

haru-a8n, 2017-05-31:
Thanks oroulet! this is now fixed.

closed 2017-05-31 by haru-a8n