FreeOpcUa/python-opcua #455: Can not connect to Softing OPC UA server with username and password, BadIdentityTokenInvalid

kbischop, 2017-06-01:
I´m using the Softing OPC UA server SDK in my application ([Softing SDK](https://industrial.softing.com/en/products/software/opc-development-toolkits/opc-ua-c-development-toolkits-platform-independent/opc-ua-c-server-client-toolkit-for-windows.html)) and try to connect with freeopcua /python-opcua 0.90.3 to this Softing OPC OPC UA server. I´m using windows 7 64 bit and python 3.5.2

I can only connect with anonymous to the OPC UA server. Whenever I try to use a username , password combination I get the error BadIdentityTokenInvalid.

I checked with other OPC UA clients -> node.js with node-opcua package, UaExpert and Softings own OPC UA client. All these clients work without problems with the anonymous account but also with the user account with username and password.

Are there known problems with the Softing OPC UA Server ?

Here is the log:

> INFO:opcua.client.ua_client.Socket:opening connection
INFO:opcua.client.ua_client.Socket:Thread started
DEBUG:opcua.uaprotocol:Waiting for header
INFO:opcua.uaprotocol:received header: Header(type:b'ACK', chunk_type:b'F', body
_size:20, channel:0)
DEBUG:opcua.uaprotocol:Waiting for header
INFO:opcua.client.ua_client.Socket:open_secure_channel
DEBUG:opcua.client.ua_client.Socket:Sending: OpenSecureChannelRequest(TypeId:Fou
rByteNodeId(i=446), RequestHeader:RequestHeader(AuthenticationToken:TwoByteNodeI
d(i=0), Timestamp:2017-06-01 09:50:26.035799, RequestHandle:1, ReturnDiagnostics
:0, AuditEntryId:None, TimeoutHint:1000, AdditionalHeader:None), Parameters:Open
SecureChannelParameters(ClientProtocolVersion:0, RequestType:SecurityTokenReques
tType.Issue, SecurityMode:MessageSecurityMode.None_, ClientNonce:b'', RequestedL
ifetime:3600000))
INFO:opcua.uaprotocol:received header: Header(type:b'OPN', chunk_type:b'F', body
_size:123, channel:10)
DEBUG:opcua.uaprotocol:Waiting for header
INFO:opcua.client.ua_client:create_session
DEBUG:opcua.client.ua_client.Socket:Sending: CreateSessionRequest(TypeId:FourByt
eNodeId(i=461), RequestHeader:RequestHeader(AuthenticationToken:TwoByteNodeId(i=
0), Timestamp:2017-06-01 09:50:26.041800, RequestHandle:2, ReturnDiagnostics:0,
AuditEntryId:None, TimeoutHint:1000, AdditionalHeader:None), Parameters:CreateSe
ssionParameters(ClientDescription:ApplicationDescription(ApplicationUri:urn:free
opcua:client, ProductUri:urn:freeopcua.github.no:client, ApplicationName:Localiz
edText(Encoding:2, Locale:None, Text:b'Pure Python Client'), ApplicationType:App
licationType.Client, GatewayServerUri:None, DiscoveryProfileUri:None, DiscoveryU
rls:[]), ServerUri:None, EndpointUrl:opc.tcp://localhost:4880, SessionName:Pure
Python Client Session1, ClientNonce:b"\xcd\x82\x0c6\x08Q\rC'\x10\x8ev/\xee\xb5b\
xc8B\x88\xe7\x1dCMd\xed\xf5d\xe5\xcd\x19\x9c\xf8", ClientCertificate:None, Reque
stedSessionTimeout:3600000, MaxResponseMessageSize:0))
INFO:opcua.uaprotocol:received header: Header(type:b'MSG', chunk_type:b'F', body
_size:573, channel:10)
DEBUG:opcua.uaprotocol:Waiting for header
DEBUG:opcua.client.ua_client:CreateSessionResponse(TypeId:FourByteNodeId(i=464),
 ResponseHeader:ResponseHeader(Timestamp:2017-06-01 09:50:26.049801, RequestHand
le:2, ServiceResult:StatusCode(Good), ServiceDiagnostics:DiagnosticInfo(Encoding
:0, SymbolicId:0, NamespaceURI:0, Locale:0, LocalizedText:0, AdditionalInfo:None
, InnerStatusCode:StatusCode(Good), InnerDiagnosticInfo:None), StringTable:None,
 AdditionalHeader:None), Parameters:CreateSessionResult(SessionId:NumericNodeId(
ns=3;i=3674407209), AuthenticationToken:ByteStringNodeId(b=b")\x01\x03\xdb\x1b\x
9d'\x8b\x04\xef\x05\x89>6p\xbf\x85Lo\xca\xb5!\x10uE\n\x8b\xb0\xec\x8e\x10\x8a"),
 RevisedSessionTimeout:600000.0, ServerNonce:b'\x0f\xe9\r\x81\xd4\x1e\x15\xe36\x
98;XZ\x8c\xa1\x8d\xf2\xbd]~0\x8d\xdcX\xa7\xaf\xf0j\xcbx\x82b', ServerCertificate
:None, ServerEndpoints:[EndpointDescription(EndpointUrl:opc.tcp://localhost:4880
/, Server:ApplicationDescription(ApplicationUri:urn:TC028L/Softing/OpcUa/TestSer
ver, ProductUri:urn:Softing/Products/OpcUa/SampleServer1, ApplicationName:Locali
zedText(Encoding:3, Locale:b'en', Text:b'Softing OpcUa Sample Server_1'), Applic
ationType:ApplicationType.ClientAndServer, GatewayServerUri:None, DiscoveryProfi
leUri:None, DiscoveryUrls:[]), ServerCertificate:None, SecurityMode:MessageSecur
ityMode.None_, SecurityPolicyUri:http://opcfoundation.org/UA/SecurityPolicy#None
, UserIdentityTokens:[UserTokenPolicy(PolicyId:Anonymous_Policy, TokenType:UserT
okenType.Anonymous, IssuedTokenType:None, IssuerEndpointUrl:None, SecurityPolicy
Uri:None), UserTokenPolicy(PolicyId:Username_Policy, TokenType:UserTokenType.Use
rName, IssuedTokenType:None, IssuerEndpointUrl:None, SecurityPolicyUri:http://op
cfoundation.org/UA/SecurityPolicy#None)], TransportProfileUri:http://opcfoundati
on.org/UA-Profile/Transport/uatcp-uasc-uabinary, SecurityLevel:11)], ServerSoftw
areCertificates:[], ServerSignature:SignatureData(Algorithm:None, Signature:b'')
, MaxRequestMessageSize:0))
DEBUG:opcua.client.client:starting keepalive thread with period of 420000.0 mill
iseconds
WARNING:opcua.client.client:Sending plain-text password
INFO:opcua.client.ua_client:activate_session
DEBUG:opcua.client.ua_client.Socket:Sending: ActivateSessionRequest(TypeId:FourB
yteNodeId(i=467), RequestHeader:RequestHeader(AuthenticationToken:ByteStringNode
Id(b=b")\x01\x03\xdb\x1b\x9d'\x8b\x04\xef\x05\x89>6p\xbf\x85Lo\xca\xb5!\x10uE\n\
x8b\xb0\xec\x8e\x10\x8a"), Timestamp:2017-06-01 09:50:26.061802, RequestHandle:3
, ReturnDiagnostics:0, AuditEntryId:None, TimeoutHint:1000, AdditionalHeader:Non
e), Parameters:ActivateSessionParameters(ClientSignature:SignatureData(Algorithm
:b'http://www.w3.org/2000/09/xmldsig#rsa-sha1', Signature:b''), ClientSoftwareCe
rtificates:[], LocaleIds:['en'], UserIdentityToken:UserNameIdentityToken(PolicyI
d:Username_Policy, UserName:xxxxxxx, Password:xxxxxxxx
derWelt4711, EncryptionAlgorithm:), UserTokenSignature:SignatureData(Algorithm:N
one, Signature:None)))
INFO:opcua.uaprotocol:received header: Header(type:b'MSG', chunk_type:b'F', body
_size:40, channel:10)
DEBUG:opcua.uaprotocol:Waiting for header
WARNING:opcua.client.ua_client.Socket:ServiceFault from server received  in resp
onse to ActivateSessionRequest
Traceback (most recent call last):
  File "OpcUaClient.py", line 81, in <module>
    test.connect()
  File "OpcUaClient.py", line 27, in connect
    self.client.connect()
  File "C:\Program Files (x86)\Python35-32\lib\site-packages\opcua\client\client
.py", line 247, in connect
    self.activate_session(username=self._username, password=self._password, cert
ificate=self.user_certificate)
  File "C:\Program Files (x86)\Python35-32\lib\site-packages\opcua\client\client
.py", line 419, in activate_session
    return self.uaclient.activate_session(params)
  File "C:\Program Files (x86)\Python35-32\lib\site-packages\opcua\client\ua_cli
ent.py", line 248, in activate_session
    data = self._uasocket.send_request(request)
  File "C:\Program Files (x86)\Python35-32\lib\site-packages\opcua\client\ua_cli
ent.py", line 77, in send_request
    self.check_answer(data, " in response to " + request.__class__.__name__)
  File "C:\Program Files (x86)\Python35-32\lib\site-packages\opcua\client\ua_cli
ent.py", line 86, in check_answer
    hdr.ServiceResult.check()
  File "C:\Program Files (x86)\Python35-32\lib\site-packages\opcua\ua\uatypes.py
", line 207, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadIdentityTokenInvalid: The user identity token is not

> valid.(BadIdentityTokenInvalid)

destogl, 2017-06-01:
Can you check with wireshark and try to find differences in Token sending?

kbischop, 2017-06-01:
I hope I got the necessary data. What do I have to look for ?

``
No.     Time           Source                Destination           Protocol Length Info
     31 10.265088      192.168.100.234       192.168.100.141       TCP      66     50169 → 4880 [SYN] Seq=0 Win=8192 Len=0 MSS=1460 WS=256 SACK_PERM=1

Frame 31: 66 bytes on wire (528 bits), 66 bytes captured (528 bits) on interface 0
Ethernet II, Src: WistronI_98:e2:37 (3c:97:0e:98:e2:37), Dst: Siemens_a3:6d:55 (00:1b:1b:a3:6d:55)
Internet Protocol Version 4, Src: 192.168.100.234, Dst: 192.168.100.141
Transmission Control Protocol, Src Port: 50169, Dst Port: 4880, Seq: 0, Len: 0

No.     Time           Source                Destination           Protocol Length Info
     32 10.265825      192.168.100.141       192.168.100.234       TCP      66     4880 → 50169 [SYN, ACK] Seq=0 Ack=1 Win=65535 Len=0 MSS=1460 WS=2 SACK_PERM=1

Frame 32: 66 bytes on wire (528 bits), 66 bytes captured (528 bits) on interface 0
Ethernet II, Src: Siemens_a3:6d:55 (00:1b:1b:a3:6d:55), Dst: WistronI_98:e2:37 (3c:97:0e:98:e2:37)
Internet Protocol Version 4, Src: 192.168.100.141, Dst: 192.168.100.234
Transmission Control Protocol, Src Port: 4880, Dst Port: 50169, Seq: 0, Ack: 1, Len: 0

No.     Time           Source                Destination           Protocol Length Info
     33 10.266349      192.168.100.234       192.168.100.141       TCP      54     50169 → 4880 [ACK] Seq=1 Ack=1 Win=65536 Len=0

Frame 33: 54 bytes on wire (432 bits), 54 bytes captured (432 bits) on interface 0
Ethernet II, Src: WistronI_98:e2:37 (3c:97:0e:98:e2:37), Dst: Siemens_a3:6d:55 (00:1b:1b:a3:6d:55)
Internet Protocol Version 4, Src: 192.168.100.234, Dst: 192.168.100.141
Transmission Control Protocol, Src Port: 50169, Dst Port: 4880, Seq: 1, Ack: 1, Len: 0

No.     Time           Source                Destination           Protocol Length Info
     34 10.269335      192.168.100.234       192.168.100.141       OpcUa    116    Hello message

Frame 34: 116 bytes on wire (928 bits), 116 bytes captured (928 bits) on interface 0
Ethernet II, Src: WistronI_98:e2:37 (3c:97:0e:98:e2:37), Dst: Siemens_a3:6d:55 (00:1b:1b:a3:6d:55)
Internet Protocol Version 4, Src: 192.168.100.234, Dst: 192.168.100.141
Transmission Control Protocol, Src Port: 50169, Dst Port: 4880, Seq: 1, Ack: 1, Len: 62
OpcUa Binary Protocol
    Message Type: HEL
    Chunk Type: F
    Message Size: 62
    Version: 0
    ReceiveBufferSize: 65536
    SendBufferSize: 65536
    MaxMessageSize: 0
    MaxChunkCount: 0
    EndPointUrl: opc.tcp://192.168.100.141:4880

No.     Time           Source                Destination           Protocol Length Info
     35 10.269808      192.168.100.141       192.168.100.234       OpcUa    82     Acknowledge message

Frame 35: 82 bytes on wire (656 bits), 82 bytes captured (656 bits) on interface 0
Ethernet II, Src: Siemens_a3:6d:55 (00:1b:1b:a3:6d:55), Dst: WistronI_98:e2:37 (3c:97:0e:98:e2:37)
Internet Protocol Version 4, Src: 192.168.100.141, Dst: 192.168.100.234
Transmission Control Protocol, Src Port: 4880, Dst Port: 50169, Seq: 1, Ack: 63, Len: 28
OpcUa Binary Protocol
    Message Type: ACK
    Chunk Type: F
    Message Size: 28
    Version: 0
    ReceiveBufferSize: 65536
    SendBufferSize: 65536
    MaxMessageSize: 16777216
    MaxChunkCount: 0

No.     Time           Source                Destination           Protocol Length Info
     36 10.275734      192.168.100.234       192.168.100.141       OpcUa    186    OpenSecureChannel message: OpenSecureChannelRequest

Frame 36: 186 bytes on wire (1488 bits), 186 bytes captured (1488 bits) on interface 0
Ethernet II, Src: WistronI_98:e2:37 (3c:97:0e:98:e2:37), Dst: Siemens_a3:6d:55 (00:1b:1b:a3:6d:55)
Internet Protocol Version 4, Src: 192.168.100.234, Dst: 192.168.100.141
Transmission Control Protocol, Src Port: 50169, Dst Port: 4880, Seq: 63, Ack: 29, Len: 132
OpcUa Binary Protocol
[... 141 lines cut ...]
    SecureChannelId: 22
    Security Token Id: 1
    Security Sequence Number: 3
    Security RequestId: 3
    OpcUa Service : Encodeable Object
        TypeId : ExpandedNodeId
            NodeId EncodingMask: Four byte encoded Numeric (0x01)
            NodeId Namespace Index: 0
            NodeId Identifier Numeric: ServiceFault (397)
        ServiceFault
            ResponseHeader: ResponseHeader
                Timestamp: Jun  1, 2017 15:26:47.923389100 Mitteleuropäische Sommerzeit
                RequestHandle: 3
                ServiceResult: 0x80200000 [BadIdentityTokenInvalid]
                ServiceDiagnostics: DiagnosticInfo
                    EncodingMask: 0x00
                StringTable: Array of String
                    ArraySize: -1
                AdditionalHeader: ExtensionObject
                    TypeId: ExpandedNodeId
                    EncodingMask: 0x00

No.     Time           Source                Destination           Protocol Length Info
     42 10.504316      192.168.100.234       192.168.100.141       TCP      54     50169 → 4880 [ACK] Seq=728 Ack=776 Win=64768 `Len=0
``

zerox1212, 2017-06-01:
The last person that had this issue said that the cause was a corporate laptop with some security stuff blocking opc ua.

kbischop, 2017-06-01:
Well this was my first thought that some security stuff is blocking ops ua, but as wrote I also checked with node.js and the node-opcua package, the Ua Expert opc ua client Softings opc ua client and a c application which uses the open62541 library. All these clients can connect using username and password. Only my python app with freeopcua cannot connect. I even tested with opcua-client-gui which also uses freeopcua and it can not connect.  I tested with the client installed on one computer and the server on the other computer and also both client and server on the same computer, but this makes no difference.

Any further ideas what I can check ?

oroulet, 2017-06-01:
Hi, with wireshark take a falling log session with python-opcua and a
working one with uaexpert then compare them. If you do not manage to spot
the difference in authentication yourself, upload them to this bug report

oroulet, 2017-06-01:
The difference should be somewhere in the last packet we send before the
error

mirkocomparetti-synesis, 2021-04-15:
An old one, but I'm facing the same issue. Did anyone manage to find a solution?

Thanks