FreeOpcUa/python-opcua #456: unable to deserialize key with pem certificates

mattfield11, 2017-06-08:
When I load pem certificates using load_private_key, I am getting error:

``` File "/home/jmm/workspace/virtualenv/zib3/lib/python3.5/site-packages/opcua/crypto/uacrypto.py", line 36, in load_private_key
    return serialization.load_der_private_key(f.read(), password=None, backend=default_backend())
  File "/home/jmm/workspace/virtualenv/zib3/lib/python3.5/site-packages/cryptography/hazmat/primitives/serialization.py", line 28, in load_der_private_key
    return backend.load_der_private_key(data, password)
  File "/home/jmm/workspace/virtualenv/zib3/lib/python3.5/site-packages/cryptography/hazmat/backends/openssl/backend.py", line 1022, in load_der_private_key
    password,
  File "/home/jmm/workspace/virtualenv/zib3/lib/python3.5/site-packages/cryptography/hazmat/backends/openssl/backend.py", line 1164, in _load_key
    self._handle_key_loading_error()
  File "/home/jmm/workspace/virtualenv/zib3/lib/python3.5/site-packages/cryptography/hazmat/backends/openssl/backend.py", line 1222, in _handle_key_loading_error
    raise ValueError("Could not deserialize key data.")
```
If I look at the code in uacrypto.py I find the following:

```
def load_private_key(path):
    _, ext = os.path.splitext(path)
    with open(path, "rb") as f:
        if ext == ".pem":
            return serialization.load_pem_private_key(f.read(), password=None, backend=default_backend())
        else:
            return serialization.load_der_private_key(f.read(), password=None, backend=default_backend())
```

It seems to me that this is a bug, because a pem private key will often have an extension that is not .pem.  (for example .key)  My work around is to rename my key with a pem extension, but wouldn it be better to explicitly define the certificate encoding when calling the function as pem or der rather than relying on the extension.?(we often find certificates with .crt or cer suffix).. Thankyou

oroulet, 2017-06-08:
God idea. Send a pr

mattfield11, 2017-06-13:
OK, I will have a go once i finish my testing in the next few weeks. Thanks

deoresheetal, 2017-09-12:
Any updates on this?

mattfield11, 2017-09-12:
No, I am afraid my development has taken me in another direction so i
haven't got round to it. I worked around by renaming my key files.