FreeOpcUa/python-opcua #459: separate setup routines server client for ARM processor cross compile

andresbh, 2017-06-21:
I would like to use the python OPC-UA client on a ARM board but compiling the complete application is too much for the small board running fedora.

I would appreciate it if separate setup.py for client and server could be made available.

zerox1212, 2017-06-21:
Just git pull the library from here and delete the client or server folder. However I don't think it will save you much. Is your problem that disk space or ram?

andresbh, 2017-06-21:
combination of ram 0.5 Gb and processor.

zerox1212, 2017-06-21:
Are you loading the address space from a cache file? The UA stack is rather large. I'm not sure if 512 of ram is enough.

oroulet, 2017-06-21:
as @zerox1212 wrote. splitting client and server will not help you. It will only save disk place. You should try disabling hte loading of the adress space, this is probably what is taking time. or use cache. If it does not work you need to use the c stack, search for it in github