FreeOpcUa/python-opcua #460: BadSecurityChecksFailed

carlosgs91, 2017-06-23:
Hi,

I am trying to connect to a demo server from KEPServerEX. I can do so with OpenOPC by using

```
opc = OpenOPC.open_client('localhost')
opc.connect('Kepware.KEPServerEX.V6')
```
To do the same in FreeOPCUa I have tried using:

```
client = Client("opc.tpc://127.0.0.1:49320")
client.connect()
```

And also:

```
client = Client("opc.tpc://127.0.0.1:49320/Kepware.KEPServerEX.V6")
client.connect()
```

But I get the following error:

[image]

I am not sure if it's related but I don't have cryptography installed.

Any idea what's the issue?

oroulet, 2017-06-23:
You were denied access. You need to find out what your server want. Maybe
it wants encryption...

AndrewStuhr, 2020-03-03:
I'm also receiving BadSecurityChecksFailed while trying to connect to Kepware.  I do have cryptography installed.  Kepware is on the PC running FreeOpcUa.  Kepware does not require any user credentials on this PC.  Any thoughts/tips for fixing the issue?  Thanks!

AndrewStuhr, 2020-03-03:
Ok I figured it out.  Kepware requires an encryption certificate by default, and I wasn't providing one.  Eventually I may use a certificate, but for the time being, I disabled Kepware security by opening OPC UA Configuration  >  click on server endpoint  >  select "None" for security policy