FreeOpcUa/python-opcua #462: python-opcua Supported Features Documentation

aquari89, 2017-07-03:
Hello, All.

I looked at other opcua opensource and saw that the features were very well documented.
However, the python-opc ua was only listed as a sentence.
From the point of view of using python-opcua, I thought that it would be nice to show the functions by table.

ex) open62541, node-opcua
[image]
https://github.com/open62541/open62541/blob/0.2/FEATURES.md
[image]
https://github.com/node-opcua/node-opcua

oroulet, 2017-07-03:
Can make/copy list at end of readme ? and I will go over and check for
errors / fill up what is missing

aquari89, 2017-07-03:
@oroulet 
I made a table with the list in node-opcua.
`https://github.com/aquari89/python-opcua/tree/patch-1`
Compared to open62541, it is almost similar, but there are a few different lists.

I would appreciate any corrections to the freeopcua by referring to the link above.
If there is anything I can do, tell me.
I will help you as much as I can.

oroulet, 2017-07-03:
Great, thanks for your help. Maybe add a second column in first table so we have one column for server and one for client? Can you also star filling up what you know so I understand how you want to fill it up?
A comment would also be usefull in first table, and we should also have a column for client and one for server in second table. The status might be different

oroulet, 2017-07-03:
let me know when you are ready for more input

aquari89, 2017-07-04:
@oroulet 
Make sure the table is configured as you said.
`https://github.com/aquari89/python-opcua/blob/patch-1/README.md`

Tell me about the part that needs to be fixed (like a service list or something else).
If we talk to each other in words, we may not know each other 's intentions. If you can organize the tables and lists with a document like Excel, I will update you as soon as possible. (I attached excel file.)
[pythonopcua_features_table.xlsx](https://github.com/FreeOpcUa/python-opcua/files/1120855/pythonopcua_features_table.xlsx)

And
Regarding display ...
If the implementation is done and is included in the release version 
[image]

If your implementation is complete, but you plan to include it in the next release
[image]

Next, depending on the degree of implementation, I would like to mark it as follows.
[image]
[image]
[image]

And if you do not have an implementation plan yet, you can leave it blank.

aquari89, 2017-07-04:
I do not know about the status indication.
However, I guessed it by referring to open62541 or node-opcua.
Tell me if you need to change the status display.

oroulet, 2017-07-04:
btw what about having this tables in a wiki and not directly in README? so that anyone can work on it?

oroulet, 2017-07-04:
https://github.com/FreeOpcUa/python-opcua/wiki/python-opcua-Supported-Features

just an idea. We would then need to add a link from README

oroulet, 2017-07-04:
I am fine with the resulting layout, but  in the code we should try to make it more compact, when editing wiki, every line sans over 2 lines and it makes things difficult to edit

aquari89, 2017-07-05:
It's also a good idea to have 'support features' on the wiki.
However, as you said, if you want to modify the table compactly in the wiki, we'll have to change the composition of the table.

How do you create a new 'Features.md' instead of 'Read.md' like open62541?
[image]

I hope that developers using pythonopcua will have a good understanding of what pythonopcua currently supports.

oroulet, 2017-07-05:
No need to make one more file to repository. I think we put the list in wiki. Work on that one: https://github.com/FreeOpcUa/python-opcua/wiki/python-opcua-Supported-Features
so I can update the README later to point to it

oroulet, 2017-07-05:
It looks good. We just need to find a way to remove all these empty spaces in markdown so lines do not spawn over 2 lines

aquari89, 2017-07-06:
In the first "Service" table, I modified `XXXX Service Set -> XXXX`.
ex)
[image]