FreeOpcUa/python-opcua #466: Get Item by BrowseName

Henna1977, 2017-07-05:
Hello,
I'm not realy sure if this is an Issue.
But I have list of some Browsenames of a group of items (100+)

I want to put all of this items into a subscription to walk along the browsename takes a long time, because some nodes have more ten 1000 childenodes.

Is there a good way to get the node by using the browsename or do I have to build my own fast method with caching?

Best regards
Henna

oroulet, 2017-07-05:
Get_child() ? but I am not sure I understand the question

zerox1212, 2017-07-05:
Are you saying that you are looping through your address space and comparing to browsename to find a child? Or are you trying to subscribe to over ten thousand nodes?

Henna1977, 2017-07-06:
Hello,

get_child() was real good tip, but I got a small problem.

The structure of the OPC Server i work with is like:

Root +
         |+ Objects +
                            |- Server (with default entries)
                            |- MainNode1 
                            |- MainNode2 +
                                                   |- DataFolder1 (with 3000+ Items)
                                                   |- DataFolder2 (with 3000+ Items)
                                                   |- DataFolder3 (with 3000+ Items)
                                                   |- DataFolder4 (with 3000+ Items)
                            |- MainNode3
                            |- MainNode4   

All main nodes have nearly the same structure.
So in sum there are a lot of items available.

Now I need to subscribe to some items given by a list of path names like this:

Root\Objects\MainNode2\DataFolder1\Dataitem_302
Root\Objects\MainNode2\DataFolder1\Dataitem_507
Root\Objects\MainNode2\DataFolder1\Dataitem_823
Root\Objects\MainNode2\DataFolder1\Dataitem_1010
.....

When I work with the function get_child() on the root node I can pass a list like
['Objects', 'Server', 'ServerStatus', 'CurrentTime']
and I receive the Nodeobject .

No Problem so far, but when I pass a list like 
['Objects', 'MainNode2', 'DataFolder1', 'Dataitem_302']
no node will be found. I think the reason is the browseName of the single nodes, because the browsename of MainNode2 is for example MainNode2 is "4: MainNode2"

If I modify the list of the path like this:
['Objects', '4: MainNode2', '4: DataFolder1', '4: Dataitem_302']
I receive the expected node.
The Problem is I just get the path information but not the Namespace Index.

I hope this explains my problem.

oroulet, 2017-07-06:
Namespace is an essential part of protocol. You need it! But probably it is
same namespace everywhere

zerox1212, 2017-07-06:
> When I work with the function get_child() on the root node I can pass a list like
['Objects', 'Server', 'ServerStatus', 'CurrentTime']
and I receive the Nodeobject .

This must mean that `get_child()' defaults to namespace 0?

oroulet, 2017-07-07:
Btw if you have many nodes you can query for all of them in one call. Look at code in get_child()

Henna1977, 2017-07-07:
Thanks for the answers.

I will take a look on the code of the get_child() method.

But for now I have a solution to automaticly find the correct namespace index if I have to change it for a child node.

@zerox1212: yes, it looks like the default namespace is 0. In my case every MainNode has it own namespace

closed 2017-07-07 by Henna1977