FreeOpcUa/python-opcua #467: Confirm that creating monitored items or subscriptions

aquari89, 2017-07-08:
Hello, All.

Among the server functions,
Is the ability for the server to verify that client tools(ex: UaExpert) have created monitored items or subscriptions?

If the python-opcua server part has that capability, can you tell me what it is?

zerox1212, 2017-07-08:
Do you want a list of clients, or a list of nodes that have subscriptions?

aquari89, 2017-07-09:
I want to know a list of nodes that have subscriptions.

Can I know if a particular node has a subscription, not an entire node list?

zerox1212, 2017-07-09:
I would have to look or you should wait for @oroulet . As far as I know there is no method on the node for checking this. You would probably have to dig into the server's session object or maybe the internal address space dict.

destogl, 2017-07-10:
this is stored in internal address space dict. Theoretically is possible and should be relatively easy to implement it.

@aquari89 how would look the call to the function to check this?

oroulet, 2017-07-11:
Knowing if a node has subscription is cheap, knowing which subscription is harder. Knowing which client need an api change

aquari89, 2017-07-13:
@zerox1212 @destogl @oroulet 
Thank you for your reply!

destogl, 2019-07-31:
This seams to be solved.

closed 2019-07-31 by destogl