FreeOpcUa/python-opcua #471: XML importer - missing methods

XB32Z, 2017-07-12:
Hi all, 

I made a model that have an object type that I instantiated (also in an XML). The type has a folder that contains a few methods. However when I import it both Type and instances don't have the methods. Only one method is present because it doesn't have any input our output arguments. I suppose that it might be a design issue however my model works fine in C/C++ using [open62541](http://github.com/open62541/open62541). 
Any ideas why the methods are missing?

Additional info: I read the Debug information and there's nothing special, it write that the method was imported. I also tried to browse to it but it is simply not there. 

Thx

oroulet, 2017-07-12:
Do the open62541 import xml now? but anyway we need more informatino in order to debug. a SIMPLE exaple xml that breaks for example

XB32Z, 2017-07-12:
Hi, 

No sorry I meant not imported but compiled. 
However by trying to reproduce a simple test that makes the error I realized that actually the problem is that the problems isn't related to folders but to references. When I reference a method in another node, it disappear from the node from which it is children. I am finishing the buggy xml and post it

XB32Z, 2017-07-12:
Ok sorry for all the posts, I followed the debug until I figured that:
 It happens for all nodes (methods or objects or variables) that are referenced with a custom ReferenceType
 I followed it and it turns out that the target node is not in the address space when the reference is read. Any idea why?

oroulet, 2017-07-12:
Can send a minimal example?

XB32Z, 2017-07-13:
Hi, 

After many test to reproduce the error on a simple file, I chose to keep trying to debug instead. (Also I cannot give you my files). I think the issue was caused by a tight interconnection between various namespaces which couldn't be resolved when adding the reference. 

Anyway I made my way through by using add_object from the type instead of using an Instance from an XML file. However by doing so I encountered a "NoneType" error in the serializer that I solved by adding a test on NoneType. L90 of instanciate.py, the command added_nodes.extend(nodeids) can fail if nodeids is None. 

Best,