FreeOpcUa/python-opcua #472: Matlab client cannot connect to python-opcua example server

eskildsf, 2017-07-13:
My setup is:
Python on Windows executing the server-example.py script
Matlab running on Windows attempting to connect to python-opcua server using the Matlab OPC Toolbox OPC-UA client.

For the Python client I simply run: python server-example.py which yields "Listening on 0.0.0.0:4840" indicating that the server runs without issues.
In Matlab I run 
```
ua_client = opcua('opc.tcp://localhost:4840/freeopcua/server/');
connect(ua_client)
```

which yields
```
ua_client = OPC UA Client OPC UA Server at localhost on port 4840:
Hostname: localhost
Port: 4840
Timeout: 10

Status: Disconnected

Error using opc.ua.Client/connect (line 420)
Could not retrieve endpoints for server. Received message: A low level communication error occurred. Ensure that server host is reachable.
```

I have ensured that the server host is reachable by connecting to the sever with both the OPC Foundation reference client and the opcua-client offered by FreeOpcUa. Both clients can read data without any problems.

When I try to connect using Matlab by executing `connect(ua_client)` then I see activity in the OPC-UA server which reads
`Cleanup client connection: ('127.0.0.1', 50678)`
where 50678 seems to increment by 1 every time I try to connect.

I have attempted to run the OPC-UA server on a computer on LAN and observed the same problem. I captured the network traffic using Wireshark. I have attached the capture to this issue.
[Capture.pcapng.zip](https://github.com/FreeOpcUa/python-opcua/files/1144231/Capture.pcapng.zip)

I hope that this report is useful.

Best regards

oroulet, 2017-07-13:
Looks like matlab is not happy with the server description we send. Do you have another server availbable you can test with? then record the session. We can try to spot the difference in server description

eskildsf, 2017-07-13:
Hi Oroulet. I have attached a Wireshark capture between the OPC Foundation Unified Architecture Technology Samples'  Generic Server and the Matlab OPC-UA client subject to the same comamnds as previously mentioned:
 ```
ua_client = opcua('opc.tcp://192.168.100.2:51210/UA/SampleServer');
connect(ua_client)
```
Matlab connects silently indicating that everything is in order. Furthermore I am able to read variables etc. in Matlab.

I should also mention that I have received a new reply in the Google thread though I think we should consolidate the communication here.

Thank you for your assistance.

Best regards

[capture_working.pcapng.zip](https://github.com/FreeOpcUa/python-opcua/files/1144636/capture_working.pcapng.zip)

destogl, 2017-07-13:
I spotted few differences which are possible to make problems:

ApplicationUri:
urn:freeopcua:python:server | urn:DESKTOP-KN3JB4J:UA Sample Server

ProductUri:
urn:freeopcua.github.no:python:server
http://opcfoundation.org/UA/SampleServer

ApplicationType:
ClientAndServer 0x00000002
Server 0x00000000

DiscoveryURLs:
opc.tcp://0.0.0.0:4840/freeopcua/server/
opc.tcp://localhost:51210/UA/SampleServer

My guess would be that DiscoveryURL confuses Matlab....

@eskildsf could you try to use localhost insetad of '0.0.0.0' in DiscoveryURL. I just looked at code and I think you should just try to change your server URI to localhost, or since you have two computers, concrete IP-Address.

eskildsf, 2017-07-13:
Here is a Wireshark capture with the following host setting in python and likewise in Matlab
`server.set_endpoint("opc.tcp://localhost:4840/freeopcua/server/")`

I now get the following error in Matlab:
```
>> connect(a)
Error using asyncioimpl.Channel/init
Endpoint Url must be a non-empty string.

Error in asyncio.Channel (line 114)
            customProps = obj.ChannelImpl.init(options);

Error in opc.ua.Channel (line 19)
            obj@asyncio.Channel(deviceName, converterName, varargin{:});

Error in opc.ua.Client/connect (line 472)
                        clnt.AsyncChannel = opc.ua.Channel(options,
                        [Inf,0]);
```

Best regards
[capture_localhost.pcapng.zip](https://github.com/FreeOpcUa/python-opcua/files/1145132/capture_localhost.pcapng.zip)

destogl, 2017-07-13:
I don't understand the error because server responses with Entpoint Url....

eskildsf, 2017-07-13:
Me neither. Matlab offers line numbers which I can use to trace the error but I don't feel comfortable pasting the Matlab code here, publicly.
As the error-log states Matlab fails at `opc.ua.Channel(options,..`. I have put in a breakpoint which allowed me to inspect `options` during execution which turned out to have a field called `EndpointUrl` which was empty.
Manually setting `options.EndpointUrl = 'opc.tcp://localhost:4840/freeopcua/server/'`and continuing code execution actually allowed Matlab to connect. I can also read and write values.

I will look further into it and try to figure out why Matlab sets `options.EndpointUrl` to empty in the first place.

Thank you for your help so far.

zerox1212, 2017-07-13:
Maybe it's expecting a URL in for the uri.

destogl, 2017-07-13:
OK, after short analysis I saw you get one step further. so now it brakes on "GetEntpointResponse" message.

I can confirm you that in your Wireshark data server sends EndpointUrl: "opc.tcp://127.0.0.1:4840/freeopcua/server/" . 

It is possible that Matlab uses ProductUri as endpoint URI since this one looks strange compared to UA-Server: opcua.ProductUri == "urn:freeopcua.github.no:python:server" (UA-Server: opcua.ProductUri == "http://opcfoundation.org/UA/SampleServer")

eskildsf, 2017-07-13:
Matlab sets the EndpointUrl to empty in a block of code which is supposed to check the client configuration.
It checks variables called `MessageSecurityModeEnum` and `ChannelSecurityPolicyEnum` and then calls a function `getBestSecurityConfig` which returns empty for the EndpointUrl.

Should I explicitly state security settings when connecting?

Best regards

eskildsf, 2017-07-13:
I think I might have a lead.

The Matlab OPC API offers a function called `opcuaserverinfo` which can probe for available servers on a given host.

Running the command `opcuaserverinfo('localhost')` with the Python OPC-UA server yields the exact same error as reported in https://se.mathworks.com/matlabcentral/answers/322992-connect-matlab-to-opc-ua-server . A breakpoint allowed me to see that it actually does discover the python-opcua server but that it fails on the `opcuadiscoverymex` function which is compiled, hence I cannot inspect it.

Running the command `opcuaserverinfo('desktop-kn3jb4j:51210')` with the OPC Foundation server yields no errors and is successful.

I have attached Wireshark captures for both. If you have any other ideas of what I can try then don't be afraid to let me know :-).

Best regards

[opcuaserverinfo.zip](https://github.com/FreeOpcUa/python-opcua/files/1145432/opcuaserverinfo.zip)

destogl, 2017-07-14:
I think we do not have implemented discovery options yet. @oroulet ?

Can you inspect `getBestSecurityConfig` function? It would be interesting what this function wants. Maybe Matlab want to use some kind of security but you are running the server without any.

oroulet, 2017-07-15:
@destogl good work ;-) would be great to solve that bug
I cannot inspect getBestSecurityConfig but from its name and what it returns we can guess that it selects one endpoint form its security config. So we need to sport a difference between our config and theirs. I think we should focus on the 'None' security policy, because this is probably the one they want.
In the security token we offer I spot a few difference.
We set PolicyId to  a string 'anonymous' while they set it to 0... strange from the spec it should clearly be s string.
our securitypolicy ui is empty while their is set http://opcfoundation.org/UA/SecurityPolicy#None Having it empty is also OK af. Spec
They also have a certificate set
Try to modify the values with return in internal_server.py line 230 (_set_endpoints) to different things and see if it helps. Maybe try to set a certificate too

oroulet, 2017-07-15:
@destogl server discovery based on a opcua discovery server should be implemented and working. The multicast based stuff is not implemented. As far as I know multicast for that kind of things is plainly stupide, multicast only works in small networks  where you probably do not need it discovery in industrial settings

oroulet, 2017-07-15:
btw you can probbaly open a bug request at matlab. Sinde they have issues with several servers, their implementation looks pretty crappy

eskildsf, 2017-07-15:
Hello @oroulet and @destogl.

I have contacted Mathworks and they can reproduce the issue. They are looking into a solution. I will keep you up to date on any developments.

I will try to make the modifications suggested by @oroulet to internal_server.py. Unfortunately it will have to wait till Monday morning.

I will also look into getBestSecurityConfig and see if it offers any leads.

Best regards

eskildsf, 2017-07-17:
`getBestSecurityConfig` returns an empty EndpointUrl because it itself is called with an empty set of endpoints - as such the root cause must be found upstream of this function.

I tried to change `idtoken.PolicyId = 'anonymous'` to `idtoken.PolicyId = '0'` but the error `Requested 1x18446744073709551615` still occurs.

To me it seems that the problem has to be related to the Matlab function `opc.internal.opcuadiscoverymex('getEndpointConfig', discoveryUrl)` which fails with the memory limit error.

I generated a certificate using `generate_certificate.sh` and added the following to `server-example.py` file.
```
server.load_certificate("my_cert.der")
server.load_private_key("my_private_key.pem")
```
With this modification I can connect without any issues. The conclusion must be that Matlab requires a certificate. I will report this to MathWorks as well so they can fix the underlying issue.

Thank you for your help.

closed 2017-07-17 by eskildsf

fellipy, 2017-11-04:
%FNGA - I found a error in the  getUAChannelSecurityMode, follow the one corrected by me that's %working

function channelMode = getUaChannelSecurityMode(channelURI)
%getUaChannelSecurityMode Convert channel security URI(s) into an internal mode value
%   channelMode = getUaChannelSecurityBit(channelURI) converts the URI channelURI (or cell array of URIs) into an
%   internal channel security mode value.
%
%   Channel security can be one of 'http://opcfoundation.org/UA/SecurityPolicy...'
%       #None -           [1] No security
%       #Basic128Rsa15 -  [2] Medium level security
%       #BasicRsa256 -    [3] Medium to High level security
%       #Basic256Sha256 - [4] High level security

% Copyright 2015 The MathWorks, Inc.
% developed by Opti-Num Solutions (Pty) Ltd

if ischar(channelURI)
    channelURI = {channelURI};
end
% Strip out the security mode from the URI
channelMode = zeros(size(channelURI),'uint8');
[~,modeStr] = strtok(channelURI, '#');
for cI = 1:numel(channelURI)
    if ~isempty(modeStr{cI})
        switch modeStr{cI}
            case '#None'
                channelMode(cI) = uint8(1);
            case '#Basic128Rsa15'
                channelMode(cI) = uint8(2);
            case '#Basic256'
                channelMode(cI) = uint8(3);
            case '#Basic256Sha256'
                channelMode(cI) = uint8(4);
            case '#Basic256Rsa15'
                channelMode(cI) = uint8(5);            
            otherwise
                warning(message('opc:ua:General:ChannelSecurityModeUnknown', modeStr(2:end)));
        end
    end
end

RamonGrande91, 2021-03-02:
Hi, I know this is an old post, but I have been strugling with the same topic for several days. Hope you guys can give me some hints...

@eskildsf

From what I understand, MATLAB OPC UA Client never worked with "None" security policy, correct?

Then you said as a conclussion that MATLAB requires a certificate. So you basically created a certificate (.der) and a key(.pem) in MATLAB, and put those files in your Python App (OPC UA SERVER) and programatically invoke them?

What about Server Certificates? Do they also need to reside and being invokes in the MATLAB Application?

MATLAB documentation says the next, but is too ambiguous IMO.

    **Certificate: A User Certificate (in X509 standard) is used to authenticate the user. The public key of the certificate must be pre-shared with the server, and when establishing the connection the user must provide the public key, private key, and a password used to protect the private key. Clear (passwordless) private keys are not supported by OPC Toolbox.**
source: https://www.mathworks.com/help/opc/ug/opc-ua-security.html

Thanks in advance for your help and hints.

eskildsf, 2021-03-02:
Hi @RamonGrande91 

My use case was to run an OPC-UA server in Python and connect to it with the Matlab OPC-UA client. To do this, I had to generate a key-pair and add the key-pair to the Python OPC-UA server script as stated in my post 2017-07-17. That's all there is to it. No modifications on the Matlab side.

That's how it worked in my version of Matlab at the time. I don't have access to Matlab anymore so I don't think I can help you if you are using a more recent version.

Best regards

RamonGrande91, 2021-03-03:
Hey eskildsf,

By key pair you mean .der and .pem files, right?

Did you use this connection between MATLAB and PYTHON over the network? or on localhost?

What security settings did you use finally on the Matlab source code? Did you explicitly declared them?

I am using MATLAB R2019b but I wouldnt have a problem to go back to 2017 if this works :)

Thanks for the hints.