FreeOpcUa/python-opcua #476: Arburg method to download machineprogram

mparini, 2017-07-19:
Arburg is an injection molding machine company that runs OPC UA servers on their machines.
I am able to connect to the machine and browse with UAExpert.

and with python
client = opcua.Client("opc.tcp://host_computer:password@192.100.101.30:4880/Arburg")

I can get data value of node.
node=client.get_node("ns=2;i=399800")
node.get_description()
LocalizedText(Encoding:3, Locale:en, Text:read the program from the machine)

node.get_display_name()
LocalizedText(Encoding:3, Locale:en, Text:ReadProgram)

node.get_properties()
[Node(NumericNodeId(ns=2;i=751160)), Node(NumericNodeId(ns=2;i=751161))]

With Unified Automation UaExpert I can use this method to read program from machine.

With python-opcua how can I use methods?

Massimo

oroulet, 2017-07-20:
you have an example there: https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py#L83

mparini, 2017-07-20:
Thanks,
I see the example.

but:
client=opcua.Client("opc.tcp://host_computer:marca@192.100.101.30:4880/Arburg")
client.connect()
obj = root.get_child(["0:Objects", "2:ARBURG", "2:ProgramAdministration","2:ReadProgram"])
obj.call_method()

Traceback (most recent call last):
  File "<pyshell#52>", line 1, in <module>
    obj.call_method()
TypeError: call_method() takes at least 2 arguments (1 given)

if I call:
obj.call_method(0)

Traceback (most recent call last):
  File "<pyshell#53>", line 1, in <module>
    obj.call_method(0)
AttributeError: 'int' object has no attribute 'to_binary'

where is my error?

this is in UAexpert
[image]

[image]

When I click Call
[image]

zerox1212, 2017-07-20:
Try passing an UA integer. `ua.Variant(0, ua.VariantType.Double))`

mparini, 2017-07-20:
Solution:
from opcua import ua
from opcua import Client
client=Client("opc.tcp://host_computer:marca@192.100.101.30:4880/Arburg")
client.connect()
root=client.get_root_node()
methods = root.get_child(["0:Objects", "2:ARBURG", "2:ProgramAdministration"])
method  = root.get_child(["0:Objects", "2:ARBURG", "2:ProgramAdministration","2:ReadProgram"])
result=methods.call_method(method,ua.Variant(0, ua.VariantType.UInt16))
client.disconnect()

result[0]=file name
result[1]=file program zip

Thanks.

closed 2017-07-20 by mparini