FreeOpcUa/python-opcua #477: BadUserAccessDenied on client add_variable

Avernial, 2017-07-24:
I get `BadUserAccessDenied` when I try to add variable from client to python server(server-example.py):

```
from opcua import Client

cl = Client("opc.tcp://localhost:4840/freeopcua/server")
cl.connect()
root = cl.get_root_node()
var = root.get_child(["0:Objects", "2:MyObject", "2:MyVariable"])
print(var.get_browse_name().Name)
var.add_variable(2, "test", 10)
```

Traceback:
`BadUserAccessDenied: User does not have permission to perform the requested operation.(BadUserAccessDenied)`

But this code works fine with C++ server(example_server.cpp).

How to give a client necessary permissions in python?

destogl, 2017-07-24:
Did you started server with login option?

oroulet, 2017-07-24:
The python server does not allow clients to do anything. You need to enable
remote admin and log in with admin user

zerox1212, 2017-07-24:
Should probably clarify that on the readme.

>Client: what works:
>
>connection to server, opening channel, session
>browsing and reading attributes value
>getting nodes by path and nodeids
>creating subscriptions
>subscribing to items for data change
>subscribing to events
>adding nodes **(via remote admin)**

oroulet, 2017-07-24:
Ok send a pr

Avernial, 2017-07-25:
I used a default servers from examples without login option. It is interesting that c++ allow adding nodes without remote admin.

Avernial, 2017-07-26:
Thanks for the answers.

closed 2017-07-26 by Avernial

oroulet, 2017-07-26:
It is a weakness that the c++ server allows creating nodes from clients.
User support is missing...