FreeOpcUa/python-opcua #478: Unknown Value KEPServerEX

QuentinJ2S, 2017-07-28:
Hello everyone,
I'm trying to connect a FreeOPCUA Server to the Kepware quick client.

My server works fine with UAExpert but Kepware is unable to read values. It can read tags name and type.

In quick client quality is bad and the value is reported as unknown in one menu and BAD VARTYPE in the other menu.

Does someone have an idee why Kepware is the only client that doesn't read properly my values ?

Thanks for your help

oroulet, 2017-07-28:
No idea, maybe search old bugs. I kinda remember that this client is very
bad

tennfy, 2017-07-31:
I also met this problem, uaexpert works well but kepware gets unknown value.

tvvbbb, 2018-11-19:
I meet the same problem. How to solve it.

QuentinJ2S, 2018-11-21:
Hello,
It was a long time ago but I remember adding try finally at some point and it made a difference on one of my issues.

try
      <my server code>
finally:
      server.close()

tvvbbb, 2018-11-21:
My code like this. But it doesn't work.

oroulet, 2018-11-21:
usually the issue is that the client expects SourceTimestamp or ServerTimestamp or does not support one of these

tvvbbb, 2018-11-21:
How to solve it?