FreeOpcUa/python-opcua #480: List of connected OPC UA clients

Gim6626, 2017-07-31:
Hi!

Could anybody advise me how to get list of clients connected to running server?

Thanks in advance

oroulet, 2017-07-31:
This is hard.... in the file binary_server_asyncio.py we use the the standart asyncio method 'create_server'. For some very strange reasons this method takes a class as argument which is instantiated somewhere deep in asyncio stack for every connection. I have no idea how to get the list of instantiated objects from asyncio. Maybe you should ask the asyncio developers....

Gim6626, 2017-08-01:
@oroulet thanks for comment, I've submitted issue for asyncio developers http://bugs.python.org/issue31094

oroulet, 2017-08-01:
Done in https://github.com/FreeOpcUa/python-opcua/commit/a3ac0645df3bf864a2e895983a1045a38f63f8ed

closed 2017-08-01 by oroulet