FreeOpcUa/python-opcua #481: Arburg BadIdentityTokenInvalid

mparini, 2017-08-02:
I use opcua python with 4 machine Arburg with no problem.

Now arrive new one.

I use the same 
client=opcua.Client("opc.tcp://host_computer:marca@192.101.5.30:4880/Arburg")
client=opcua.Client("opc.tcp://host_computer:marca@192.101.5.22:4880/Arburg") new one

The user name and password is the same. But for the new one we can't connect.

If I try with dataFEED or UaExpert I have no problem.
[image]

[image]

[image]

If I use uadiscover:
[image]

Any idea.
Thanks 
Massimo Parini

oroulet, 2017-08-02:
Looks like you need to for wireshark and record a session from uaexpert and one from python. Then we can try to spot a difference..

mparini, 2017-08-02:
I try. The different maybe is uaexpert and Datafeed send a certificate. The comunication is more complex that python.

[pcapng.zip](https://github.com/FreeOpcUa/python-opcua/files/1194731/pcapng.zip)

Massimo

mparini, 2017-08-03:
I find the problem:
params.UserIdentityToken.EncryptionAlgorithm = ''
in
params.UserIdentityToken.EncryptionAlgorithm = None

in opcua/client/client.py
change:
  def _add_user_auth(self, params, username, password):
        params.UserIdentityToken = ua.UserNameIdentityToken()
        params.UserIdentityToken.UserName = username
        policy_uri = self.server_policy_uri(ua.UserTokenType.UserName)
        if not policy_uri or policy_uri == security_policies.POLICY_NONE_URI:
            # see specs part 4, 7.36.3: if the token is NOT encrypted,
            # then the password only contains UTF-8 encoded password
            # and EncryptionAlgorithm is null
            if self.server_url.password:
                self.logger.warning("Sending plain-text password")
                params.UserIdentityToken.Password = password
            **params.UserIdentityToken.EncryptionAlgorithm = None #''
        elif self.server_url.password:
            data, uri = self._encrypt_password(password, policy_uri)
            params.UserIdentityToken.Password = data 
            params.UserIdentityToken.EncryptionAlgorithm = uri
        params.UserIdentityToken.PolicyId = self.server_policy_id(ua.UserTokenType.UserName, b"username_basic256")

Thanks
Massimo Parini

oroulet, 2017-08-03:
Are you making a PR? Otherwise leave bug open. If there is a bug we need to fix it

closed 2017-08-03 by mparini