FreeOpcUa/python-opcua #483: Dependency Issues

bullvai, 2017-08-02:
Hello! I was running an older (8 months ago maybe) version of python-freeopcua and I was having memory leak issues in places where there was spotty wifi connectivity. I just downloaded the newest version because I recall seeing something about a dereferencing issue being fixed, and also it looks like the shelffile has been fixed. However, when I try to run the server-example.py I get a complaint about missing "Objectify." This got me thinking about what dependencies are needed, and what I would need to download and install to get FreeOpcUa up and running using python3 from scratch. Could someone please help my out with my Objectify issue and also provide dependency information for python-opcua?

oroulet, 2017-08-02:
Looks like you are missing lxml but this should be listed as a dependent in setup.py and readme

bullvai, 2017-08-02:
Aw man, I totally missed that... /facepalm

[image]

On a side note, have you heard of people having memory leak issues when wifi connectivity is spotty?

oroulet, 2017-08-02:
No I have not, but I am interested to hear if the newer version helps. Otherwise we can try to find out where the leak is (probably a list we forget to empty somewhere)

oroulet, 2017-08-02:
Client or server?

bullvai, 2017-08-02:
Server.

zerox1212, 2017-08-02:
pip should automatically install `lxml`, correct?

bullvai, 2017-08-03:
I ended up running into an issue while using pip to install lxml. I'll take a pic with my phone when I get home today - probably about 8 hours from now.

bullvai, 2017-08-04:
Here are the issues I had with pip install lxml:

[image]

[image]

zerox1212, 2017-08-04:
The problem is most likely that you are using the default python packaged with the RPi3 OS which doesn't have as many dev libs.

You probably need to do something like this:
`sudo apt-get install python-dev libxml2-dev libxslt1-dev zlib1g-dev`

bullvai, 2017-08-04:
I just installed the above-mentioned packages and I'm waiting for pip to install lxml. It's been sitting here for quite a while, but I'm hopeful:

[image]

bullvai, 2017-08-04:
Ahhh yeah - slowly but surely, making progress!

[image]

bullvai, 2017-08-04:
[image]

zerox1212, 2017-08-04:
Probably just need to run it with root permissions `sudo pip install lxml`

bullvai, 2017-08-04:
I started running it with root permissions before I left the house this morning; hopefully all will be well when I get home.

bullvai, 2017-08-04:
lxml installed successfully:

[image]

... but I'm still getting the same error:

[image]

zerox1212, 2017-08-04:
Despite what you posted, I have a feeling you installed lxml to python 2.7. It's one of the annoying things about working in RPi, everything defaults to python 2.7.

See if `sudo pip3 install lxml` works

bullvai, 2017-08-04:
Hmm okay. To get pip3 I need to sudo apt-get install python3-pip, yeah?

bullvai, 2017-08-04:
Awesome - it works! It still complains about cryptography, but I can deal with that. Thank you very much for your help. This project is awesome!

zerox1212, 2017-08-06:
You only need crypto if you are want encryption. It's safe to ignore.

mpf82, 2017-09-11:
Maybe we can get rid of the lxml requirement, see [my comment here](https://github.com/FreeOpcUa/python-opcua/issues/474#issuecomment-328466140)