FreeOpcUa/python-opcua #488: How to kill Subscrition thread?

sanlop73, 2017-08-22:
Hi:
I am creating an small script to Read and Subscribe in the same script.
Read is a loop and Subscribe is define as the example(event loop). 
What is the way to kill subscription loop?.
Ctrl-C  not help. Just Stop the read loop.
The only way is Ctrl-Z or Crtl-\ , but unfortunately the script is still running in the background.

I want to use PM2 (node js) as a process manager to restart/shutdown  the application completelly.
Thanks a lot for your suggestions.

zerox1212, 2017-08-22:
https://github.com/FreeOpcUa/python-opcua/issues/306

sanlop73, 2017-08-22:
Hi.
Problem is client script., 
My server is an Ignition OPC server

zerox1212, 2017-08-22:
I don't know much about the client as I mainly use the server, but I assume you can stop the client as well.

oroulet, 2017-08-23:
Aa i wrote in the other bug report: I use ctrl-c all the time so this definitely works with and without subscriptions. If you have an issue it must something with your environment/code. In fact client side, a subscription does not create thread or anything, it is just a dict to remember what to do when a message from server arrives

sanlop73, 2017-08-23:
Thanks for your reply.
Is very strange, i am doing the same but using Node Js and i am not getting any issue with subscriptions.
I m  going to check again the code.
Is possible for you to check my code if i post here?

Thanks

harua8n, 2017-12-29:
Have tried typing "exit()"? This is how I exit from client-events.py.

sanlop73, 2018-01-04:
Hi:
Changed to Python 3.6 and no issues.
Thanks a lot !!